ethz-adrl/control-toolbox #13: Upgrade to newest version of HPIPM/Blasfeo

markusgft, 2019-05-12:
* in oder to avoid the issue that ct-v2 hpipm and blasfeo versions do not provide an installation routine. Exporting dependencies is therefore not solid.

jcarius, 2019-05-15:
@rubengrandia might be of help here

markusgft, 2019-05-17:
> @rubengrandia might be of help here

@rubengrandia would you be interested in committing this feature? It would be highly welcomed and would be a great contribution.

rubengrandia, 2019-05-17:
Updating to the lastest hpipm + blasfeo is still something I want to do and updating it in the CT would be a good first step. However, it is unclear when I will have time for this.

With licensing I am less familiar and is probably more relevant for you and Michael.

markusgft, 2019-10-02:
I have gotten the newest hpipm-release to work with the CT. I will commit this soon and notify everyone.

closed 2019-10-05 by markusgft