ethz-adrl/control-toolbox #32: Incompatible cppad cg code for clang++

markusgft, 2019-05-18:
@jcarius had reported the following:

Some configurations* of Clang++ cannot compile parts of the cppad cg code:

`/home/USERNAME/catkin_ws/src/ct/ct_core/include/external/cppad/cg/patterns/iter_equation_group.hpp:253:68: error: default initialization of an object of const type 'const std::map<size_t, std::set<size_t> >' (aka 'const map<unsigned long, set<unsigned long> >') without a user-provided default constructor const std::map<size_t, std::set<size_t> > IterEquationGroup<Base>::EMPTYMAPSETS;
`
The issue is summarized in this [Stackoverflow Question](https://stackoverflow.com/questions/21900237/do-i-really-need-to-implement-user-provided-constructor-for-const-objects)

(*) Yet unclear what causes this, on another machine clang worked fine.

tama10111, 2019-11-24:
Hi, I'm facing the same problem. Did you find out any solution ?

markusgft, 2019-11-24:
No, as of now, no solution has been reported. It may well be that newer releases of CppAD and/or clang++ and g++ do not show this bug, but this needs to checked and tested. 
By the way, the upcoming 3.0.2 makes CppAD and CppADCG optional, which will make it easier to circumvent this problem.