ethz-adrl/control-toolbox #58: Multiple compilation errors

tama10111, 2019-11-24:
Hello

I'm facing several compilation errors :

`/usr/local/include/ct/external/cppad/cg/patterns/index/index_pattern_impl.hpp:41:60: error: no matching function for call to ‘max(long unsigned int, std::map<unsigned int, unsigned int>::size_type)’
     size_t maxCount = std::min(std::max(3ul, x2y.size() / 4), 8ul);
`

`/usr/local/include/ct/external/cppad/local/pod_vector.hpp:58:25: error: redefinition of ‘bool CppAD::is_pod() [with Type = unsigned int]’
 template <> inline bool is_pod<size_t>(void)             { return true; }
`

`/usr/local/include/ct/external/cppad/cg/model/functor_generic_model.hpp:439:35: error: invalid conversion from ‘size_t*’ {aka ‘unsigned int*’} to ‘long unsigned int*’ [-fpermissive]
             (*_forwardOneSparsity)(j, &pos, &nnz);
`

Here is my g++ version
`ros@raspi:~/control-toolbox/ct $ g++ --version
g++ (Raspbian 8.3.0-6+rpi1) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
`
I don't really know why it doesn't works for me but works for other... Is there a specific version of g++ of some flags to give him ?

Thanks in advance

markusgft, 2019-11-24:
Dear @tama10111 
It seems like you are trying to build the CT on a Raspberry-pi device, which has in fact never been tested before. However, it should, in principle, be possible.
I successfully tested the build with gcc versions 5.4 and 7.4, version 8.3 is not tested as of now. 
You could try running that older gcc version, for instance. Another thing you can try is checking out the devel-branch of the CT, which deactivates the CppAD-functions by default. This might resolve your issues, but will limit the functionality of the library to some extent.
Best

tama10111, 2019-11-28:
Ok, thank you very much for your detailed informations :+1:

closed 2019-12-02 by markusgft