ethz-adrl/control-toolbox #59: Python Bindings

acxz, 2019-11-27:
First off, great job on this project and thank you for developing it! It pleases me to see such a toolbox available openly.

I was wondering if there are plans for creating python bindings for this C++ project. It would increase the appeal of this project and increase its ease of use. Using something like https://github.com/pybind/pybind11 would go a long way.

markusgft, 2019-11-27:
Python Bindings are on the bucket list, and pybind11 would be our favoured solution, too.
However, due to time constraints, we won't be able to offer you the feature immediately.

acxz, 2019-11-27:
I see. I am glad it is on the plan, however. I am not familiar with pybind11 myself, but I plan to learn how to use it. When I do, I would love to help integrating pybind11 support in this project.

markusgft, 2020-06-22:
#146 is a precondition for implementing python bindings.