ethz-adrl/control-toolbox #62: Forward hpipm interface to release 0.1.2 (upcoming)

markusgft, 2019-12-22:
Forward hpipm interface to release 0.1.2 (upcoming)

CedricPerottet, 2020-04-26:
Hello, and thank you for this great library !
I issue some troubles (illegal instruction) when running NLOC_boxConstrained when calling d_ocp_qp_ipm_solve (hpipm interface). 
I installed hpipm through the shell scrip provided that uses hpipm 0.1.2 commit. I wonder if the issue I experience may be related to this opened issue ?

Thank you,
Cédric

markusgft, 2020-04-26:
Dear @CedricPerottet 
thank you for reporting your issue! On which architecture and operating system are you?
Did you have any previous versions of hpipm or blasfeo installed?
Would you mind verifying if you can build the run the hpipm unit tests? For that purpose, you should build hpipm manually using cmake, using the additional test target!
Best

CedricPerottet, 2020-04-27:
I'm running on Ubuntu 18.04.4, with a 8 years old i7 core (Intel(R) Core(TM) i7-3610QM CPU @ 2.30GHz).
I did build hpipm tests with CMake as you suggested, two class member have not been updated in the tests. But more importantly, when running the tests, I get the same error as when I used it from the CT, so it seems the issue I have is not related to the CT. I will seek some help to make hpipm work first, thank you for your advice !

CedricPerottet, 2020-04-27:
I finally got hpipm to work ! 
When reading more in details how to install Blasfeo, I found that the target architecture needed to be specified in Makefile.rule as explained here : [https://blasfeo.syscop.de/docs/install/#configuration](url).
So maybe I will suggest to add a comment in the install file of hpipm in a future pull request.
Regards, Cédric

markusgft, 2020-04-27:
Dear @CedricPerottet, thanks for pointing out the issue! It would be fantastic if you would like to add a comment to the installation instructions! 
Best!

closed 2020-06-22 by markusgft