ethz-adrl/control-toolbox #63: Compiling with Cmake

KarimHassanieh, 2019-12-24:
Currently im trying to compille my code external workspace using CMake adding ct library. But I keep getting the below error :
`In file included from /usr/local/include/ct/core/core.h:18:0,
                 from simple_oscillator.cpp:1:
/usr/local/include/cppad/example/cppad_eigen.hpp:16:11: fatal error: Eigen/Core: No such file or directory
 # include <Eigen/Core>
           ^~~~~~~~~~~~
`
Is there a template CMake that can be used to use ct_core library externally?

markusgft, 2019-12-25:
Hello @KarimHassanieh 
did you make sure to install all dependencies using the _./install_deps.sh_ script from the folder _control_toolbox/ct_ ? It seems like the Eigen library is missing on your system. Or it is has been installed in a place where it cannot get located properly (I am wondering what you mean with "code external workspace")

KarimHassanieh, 2019-12-25:
Hi thanks for the feedback it compiles and runs  fine within the ct workspace but if I want develop and compile using cmake outside ct workspace which I clone from gtihub I get the error above.

markusgft, 2019-12-26:
Would you mind posting a minimal example of how you include the CT in your project using cmake? It would be interesting to reproduce the issue.

KarimHassanieh, 2019-12-27:
Hi I managed to completely resolve the issue by adding  Eigen3::Eigen to target link libraries in CMake. Anyhow Im sharing the cmake sample used just for reference in case it needs to be reproduced. 

[CMakeLists.txt](https://github.com/ethz-adrl/control-toolbox/files/4006198/CMakeLists.txt)

closed 2019-12-27 by KarimHassanieh

markusgft, 2019-12-27:
Did you try to also include find_package(ct_core) and then link against ct_core? That should be sufficient I believe. In your above CMakeLists.txt, find_package(ct_core) seems to be missing...