ethz-adrl/control-toolbox #65: Installation issue

parthos, 2019-12-28:
Does this codebase need a specific version of the cppad library?
When I build the repository, it fails because a file is not found:
```
In file included from ~/ws/ct_ws/src/control-toolbox/ct_models/include/ct/models/InvertedPendulum/codegen/InvertedPendulumActDynLinearizedForward.h:8:0,
                 from ~/ws/ct_ws/src/control-toolbox/ct_models/include/ct/models/InvertedPendulum/codegen/InvertedPendulumActDynLinearizedForward.cpp:6:
~/ws/ct_ws/devel/include/ct/core/core.h:15:10: fatal error: cppad/cg.hpp: No such file or directory
 #include <cppad/cg.hpp>
```

markusgft, 2019-12-28:
Dear @parthos,
this is indeed a bug, which shows only when cppad-codegen ist not installed. I am working on a solution, meanwhile, please feel free to use the script ./install_deps.sh in control_toolbox/ct to install the dependencies!

markusgft, 2019-12-28:
This PRQ should fix the issue https://github.com/ethz-adrl/control-toolbox/pull/64

parthos, 2019-12-29:
Hi Markus,
   Thanks for resolving that issue. I clones the latest repo and I'm getting a compilation error now:
```
In file included from /usr/include/eigen3/Eigen/Core:389:0,
                 from /usr/local/include/cppad/example/cppad_eigen.hpp:16,
                 from /usr/local/include/ct/core/core.h:21,
                 from /usr/local/include/ct/optcon/optcon.h:9,
                 from /home/starlord/ws/ct_ws/src/control-toolbox/ct_models/src/HyQ/codegen/HyQLinearizationCodegen.cpp:6:
/usr/include/eigen3/Eigen/src/Core/IO.h: In instantiation of ‘static int Eigen::internal::significant_decimals_default_impl<Scalar, IsInteger>::run() [with Scalar = CppAD::AD<CppAD::cg::CG<double> >; bool IsInteger = false]’:
/usr/include/eigen3/Eigen/src/Core/IO.h:183:66:   required from ‘std::ostream& Eigen::internal::print_matrix(std::ostream&, const Derived&, const Eigen::IOFormat&) [with Derived = Eigen::Matrix<CppAD::AD<CppAD::cg::CG<double> >, 1, 36, 1, 1, 36>; std::ostream = std::basic_ostream<char>]’
/usr/include/eigen3/Eigen/src/Core/IO.h:248:32:   required from ‘std::ostream& Eigen::operator<<(std::ostream&, const Eigen::DenseBase<Derived>&) [with Derived = Eigen::Transpose<const Eigen::Matrix<CppAD::AD<CppAD::cg::CG<double> >, 36, 1, 0, 36, 1> >; std::ostream = std::basic_ostream<char>]’
/usr/local/include/ct/rbd/state/RBDState.h:203:39:   required from ‘void ct::rbd::RBDState<NJOINTS, SCALAR>::fromStateVectorEulerXyz(const state_vector_euler_t&) [with long unsigned int NJOINTS = 12ul; SCALAR = CppAD::AD<CppAD::cg::CG<double> >; ct::rbd::RBDState<NJOINTS, SCALAR>::state_vector_euler_t = ct::core::StateVector<36ul, CppAD::AD<CppAD::cg::CG<double> > >]’
/home/starlord/ws/ct_ws/src/control-toolbox/ct_models/src/HyQ/codegen/helperFunctions.h:39:5:   required from ‘Eigen::Matrix<SCALAR, 18, 1> ct::models::HyQ::hyqInverseDynamics(const Eigen::Matrix<SCALAR, 54, 1>&) [with SCALAR = CppAD::AD<CppAD::cg::CG<double> >]’
/home/starlord/ws/ct_ws/src/control-toolbox/ct_models/src/HyQ/codegen/HyQLinearizationCodegen.cpp:28:54:   required from here
/usr/include/eigen3/Eigen/src/Core/IO.h:136:37: error: no matching function for call to ‘ceil(CppAD::AD<CppAD::cg::CG<double> >)’
     return cast<RealScalar,int>(ceil(-log(NumTraits<RealScalar>::epsilon())/log(RealScalar(10))));

```
I'm on ubuntu 16.04 and my gcc version is 5.4. I also used the script to install dependencies. I'll try this on a machine with ubuntu 18.04 as well later and report if i have the same issue.

markusgft, 2019-12-29:
Okay, I have an idea what this is. However I am absent until 1/1/20, you
could try to work with tag 3.0.1 meanwhile.

parthos, 2019-12-30:
Thanks, 3.0.1 seems to be working.

markusgft, 2020-01-03:
Dear @parthos I have been trying to reproduce your error on Ubuntu 16 and 18, but did not find a cause for it. Do I understand correctly that you have another version of CppAD already installed on your machine?

EdisonKun, 2020-01-10:
Dear @markusgft I meet the same problem in ubuntu16.04. I used script to install the dependencies in the ct folder. The version of the control toolbox is v3.0.2. When I change to v3.0.1,In file included from /usr/local/include/ct/rbd/robot/Dynamics.h:14:0,
                 from /usr/local/include/ct/rbd/Robot:11,
                 from /usr/local/include/ct/rbd/rbd.h:15,
                 from /home/kun/catkin_ws_dependency/src/control-toolbox/ct_models/src/HyQ/codegen/HyQLinearizationCodegen.cpp:7:
/home/kun/catkin_ws_dependency/devel/include/ct/rbd/state/JointState.h:21:7: error: redefinition of default argument for ‘class SCALAR’
 class JointState
       ^
compilation terminated due to -Wfatal-errors.
make[2]: *** [CMakeFiles/HyQLinearizationCodegen.dir/src/HyQ/codegen/HyQLinearizationCodegen.cpp.o] Error 1
make[1]: *** [CMakeFiles/HyQLinearizationCodegen.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /usr/local/include/ct/rbd/robot/Dynamics.h:14:0,
                 from /usr/local/include/ct/rbd/Robot:11,
                 from /usr/local/include/ct/rbd/rbd.h:15,
                 from /home/kun/catkin_ws_dependency/src/control-toolbox/ct_models/src/InvertedPendulum/codegen/InvertedPendulumWithActuatorCodeGen.cpp:12:
/home/kun/catkin_ws_dependency/devel/include/ct/rbd/state/JointState.h:21:7: error: redefinition of default argument for ‘class SCALAR’
 class JointState
       ^
compilation terminated due to -Wfatal-errors.
The errors becomes above.

EdisonKun, 2020-01-10:
The problem occurs when I run the command: catkin build -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true

markusgft, 2020-01-10:
@EdisonKun, thanks for reporting! Can you please let me know what your version of gcc (or clang) is, and on what system you are trying to build? Does the issue persist after doing a ``catkin clean --all` in your workspace? (In case you tried to build the software using bare cmake before, you would need to remove the installed files as well).
Do the examples from ct_core and ct_optcon run?

EdisonKun, 2020-01-10:
Dear @markusgft. Thanks for your reply. The gcc version is :
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.12)
and the system is ubuntu 16.04. I make the command ''catkin clean --all'' and A new problem occurs:
Errors     << ct_models:make /home/kun/catkin_ws_dependency/logs/ct_models/build.make.000.log                                                                                            
In file included from /usr/local/include/ct/rbd/robot/Dynamics.h:13:0,
                 from /usr/local/include/ct/rbd/Robot:11,
                 from /usr/local/include/ct/rbd/rbd.h:15,
                 from /home/kun/catkin_ws_dependency/src/control-toolbox/ct_models/src/InvertedPendulum/codegen/InvertedPendulumWithActuatorCodeGen.cpp:12:
/home/kun/catkin_ws_dependency/devel/include/ct/rbd/state/JointAcceleration.h:17:7: error: redefinition of default argument for ‘class SCALAR’
 class JointAcceleration
       ^
compilation terminated due to -Wfatal-errors.
make[2]: *** [CMakeFiles/InvertedPendulumWithActuatorCodeGen.dir/src/InvertedPendulum/codegen/InvertedPendulumWithActuatorCodeGen.cpp.o] Error 1
make[1]: *** [CMakeFiles/InvertedPendulumWithActuatorCodeGen.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /usr/local/include/ct/rbd/robot/Dynamics.h:13:0,
                 from /usr/local/include/ct/rbd/Robot:11,
                 from /usr/local/include/ct/rbd/rbd.h:15,
                 from /home/kun/catkin_ws_dependency/src/control-toolbox/ct_models/src/HyQ/codegen/HyQLinearizationCodegen.cpp:7:
/home/kun/catkin_ws_dependency/devel/include/ct/rbd/state/JointAcceleration.h:17:7: error: redefinition of default argument for ‘class SCALAR’
 class JointAcceleration
       ^
compilation terminated due to -Wfatal-errors.
I am finding a way to remove the installed files~and I will try again later~

EdisonKun, 2020-01-10:
Dear @markusgft. Thanks for your reply. The gcc version is :
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu116.04.12)
and the system is ubuntu 16.04. I make the command ''catkin clean --all'' and A new problem occurs:
``
Errors << ct_models:make /home/kun/catkin_ws_dependency/logs/ct_models/build.make.000.log
In file included from /usr/local/include/ct/rbd/robot/Dynamics.h:13:0,
from /usr/local/include/ct/rbd/Robot:11,
from /usr/local/include/ct/rbd/rbd.h:15,
from /home/kun/catkin_ws_dependency/src/control-toolbox/ct_models/src/InvertedPendulum/codegen/InvertedPendulumWithActuatorCodeGen.cpp:12:
/home/kun/catkin_ws_dependency/devel/include/ct/rbd/state/JointAcceleration.h:17:7: error: redefinition of default argument for ‘class SCALAR’
class JointAcceleration
^
compilation terminated due to -Wfatal-errors.
make[2]: *** [CMakeFiles/InvertedPendulumWithActuatorCodeGen.dir/src/InvertedPendulum/codegen/InvertedPendulumWithActuatorCodeGen.cpp.o] Error 1
make[1]: *** [CMakeFiles/InvertedPendulumWithActuatorCodeGen.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /usr/local/include/ct/rbd/robot/Dynamics.h:13:0,
from /usr/local/include/ct/rbd/Robot:11,
from /usr/local/include/ct/rbd/rbd.h:15,
from /home/kun/catkin_ws_dependency/src/control-toolbox/ct_models/src/HyQ/codegen/HyQLinearizationCodegen.cpp:7:
/home/kun/catkin_ws_dependency/devel/include/ct/rbd/state/JointAcceleration.h:17:7: error: redefinition of default argument for ‘class SCALAR’
class JointAcceleration
^
compilation terminated due to -Wfatal-errors.
``
I am finding a way to remove the installed filesand I will try again later~

EdisonKun, 2020-01-10:
Dear @markusgft. Thanks for your reply. The gcc version is :
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu116.04.12)
and the system is ubuntu 16.04. I make the command ''catkin clean --all'' , and the same problem exists. So I can not to run the examples from ct_core and ct_optcon, because the build error.

EdisonKun, 2020-01-10:
Dear @markusgft 
I correct the problem by remove all my compiled files, including the hpipm. But when I write the code in 
`https://ethz-adrl.github.io/ct/ct_doc/doc/html/optcon_tut_lqr.html`
I got an error like this:
`/home/kun/catkin_ws_dependency/src/quadruped_locomotion/quadruped_model/test/ct_opti/lqr_optimal.cpp:8:5: error: ‘ADCGScalar’ is not a member of ‘ct::core’
     ct::core::ADCGScalar w_n(50.0);
     ^
/home/kun/catkin_ws_dependency/src/quadruped_locomotion/quadruped_model/test/ct_opti/lqr_optimal.cpp:9:72: error: ‘ADCGScalar’ is not a member of ‘ct::core’
     std::shared_ptr<ct::core::ControlledSystem<state_dim, control_dim, ct::core::ADCGScalar>> oscillatorDynamics(
                                                                        ^
/home/kun/catkin_ws_dependency/src/quadruped_locomotion/quadruped_model/test/ct_opti/lqr_optimal.cpp:9:72: error: ‘ADCGScalar’ is not a member of ‘ct::core’
/home/kun/catkin_ws_dependency/src/quadruped_locomotion/quadruped_model/test/ct_opti/lqr_optimal.cpp:9:82: error: template argument 3 is invalid
     std::shared_ptr<ct::core::ControlledSystem<state_dim, control_dim, ct::core::ADCGScalar>> oscillatorDynamics(
                                                                                  ^
/home/kun/catkin_ws_dependency/src/quadruped_locomotion/quadruped_model/test/ct_opti/lqr_optimal.cpp:9:92: error: template argument 1 is invalid
     std::shared_ptr<ct::core::ControlledSystem<state_dim, control_dim, ct::core::ADCGScalar>> oscillatorDynamics(
                                                                                            ^
/home/kun/catkin_ws_dependency/src/quadruped_locomotion/quadruped_model/test/ct_opti/lqr_optimal.cpp:10:46: error: ‘ADCGScalar’ is not a member of ‘ct::core’
         new ct::core::tpl::SecondOrderSystem<ct::core::ADCGScalar>(w_n));
                                              ^
/home/kun/catkin_ws_dependency/src/quadruped_locomotion/quadruped_model/test/ct_opti/lqr_optimal.cpp:10:46: error: ‘ADCGScalar’ is not a member of ‘ct::core’
/home/kun/catkin_ws_dependency/src/quadruped_locomotion/quadruped_model/test/ct_opti/lqr_optimal.cpp:10:66: error: template argument 1 is invalid
         new ct::core::tpl::SecondOrderSystem<ct::core::ADCGScalar>(w_n));
                                                                  ^
/home/kun/catkin_ws_dependency/src/quadruped_locomotion/quadruped_model/test/ct_opti/lqr_optimal.cpp:10:68: error: ‘w_n’ was not declared in this scope
         new ct::core::tpl::SecondOrderSystem<ct::core::ADCGScalar>(w_n));
                                                                    ^
/home/kun/catkin_ws_dependency/src/quadruped_locomotion/quadruped_model/test/ct_opti/lqr_optimal.cpp:12:5: error: ‘ADCodegenLinearizer’ is not a member of ‘ct::core’
     ct::core::ADCodegenLinearizer<state_dim, control_dim> adLinearizer(oscillatorDynamics);
     ^
/home/kun/catkin_ws_dependency/src/quadruped_locomotion/quadruped_model/test/ct_opti/lqr_optimal.cpp:12:90: error: ‘adLinearizer’ was not declared in this scope
     ct::core::ADCodegenLinearizer<state_dim, control_dim> adLinearizer(oscillatorDynamics);
`
So I guess this caused by the cppad and CppADCodeGen. Then I install the the two libraries manually by install_acppadcg.sh file in ct folder. The I built the catkin workspace and I still get the above error.  
I guess this is caused by the installation order of ct and cppad and CppADCodeGen. So I delete all components of ct_core, ct_models, ct_rbd and ct_optcon and build them again. I got this error:
`Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
In file included from /usr/include/eigen3/Eigen/Core:389:0,
                 from /usr/local/include/cppad/example/cppad_eigen.hpp:16,
                 from /usr/local/include/ct/core/core.h:21,
                 from /usr/local/include/ct/optcon/optcon.h:9,
                 from /home/kun/system_lib/control-toolbox/ct_models/src/HyQ/codegen/HyQLinearizationCodegen.cpp:6:
/usr/include/eigen3/Eigen/src/Core/IO.h: In instantiation of ‘static int Eigen::internal::significant_decimals_default_impl<Scalar, IsInteger>::run() [with Scalar = CppAD::AD<CppAD::cg::CG<double> >; bool IsInteger = false]’:
/usr/include/eigen3/Eigen/src/Core/IO.h:183:66:   required from ‘std::ostream& Eigen::internal::print_matrix(std::ostream&, const Derived&, const Eigen::IOFormat&) [with Derived = Eigen::Matrix<CppAD::AD<CppAD::cg::CG<double> >, 1, 36, 1, 1, 36>; std::ostream = std::basic_ostream<char>]’
/usr/include/eigen3/Eigen/src/Core/IO.h:248:32:   required from ‘std::ostream& Eigen::operator<<(std::ostream&, const Eigen::DenseBase<Derived>&) [with Derived = Eigen::Transpose<const Eigen::Matrix<CppAD::AD<CppAD::cg::CG<double> >, 36, 1, 0, 36, 1> >; std::ostream = std::basic_ostream<char>]’
/usr/local/include/ct/rbd/state/RBDState.h:203:39:   required from ‘void ct::rbd::RBDState<NJOINTS, SCALAR>::fromStateVectorEulerXyz(const state_vector_euler_t&) [with long unsigned int NJOINTS = 12ul; SCALAR = CppAD::AD<CppAD::cg::CG<double> >; ct::rbd::RBDState<NJOINTS, SCALAR>::state_vector_euler_t = ct::core::StateVector<36ul, CppAD::AD<CppAD::cg::CG<double> > >]’
/home/kun/system_lib/control-toolbox/ct_models/src/HyQ/codegen/helperFunctions.h:39:5:   required from ‘Eigen::Matrix<SCALAR, 18, 1> ct::models::HyQ::hyqInverseDynamics(const Eigen::Matrix<SCALAR, 54, 1>&) [with SCALAR = CppAD::AD<CppAD::cg::CG<double> >]’
[... cut ...]
/usr/include/eigen3/Eigen/src/Core/IO.h:136:37: error: no matching function for call to ‘ceil(CppAD::AD<CppAD::cg::CG<double> >)’
     return cast<RealScalar,int>(ceil(-log(NumTraits<RealScalar>::epsilon())/log(RealScalar(10))));
                                     ^
compilation terminated due to -Wfatal-errors.`

Is this caused by the Eigen version? My eigen version is 3.2.xxx. I do not know should I upgrade it into version 3.3 because I do not know whether it should destroy other package compiling. 
Can you give me some advice on how to solve them? I have been struck by this problem about two days~Thank you so much.

EdisonKun, 2020-01-10:
Sorry, the latest problem is the same as @parthos I return back to the origin.....

markusgft, 2020-01-25:
Dear @EdisonKun @parthos 
I finally found time to track down the problem. It will be resolved shortly in pullrequest #75 
Sorry for the inconvenience!

closed 2020-01-26 by markusgft