ethz-adrl/control-toolbox #68: CT Core still has ties to non-existing "examples" folder for cppad

tsender, 2020-01-08:
I just updated to v3.0.2 and had an error compiling my catkin workspace. The CT core still tries to include files from the "cppad/examples/" folder you guys had a while back. Lines 21 and 22 in this file point to files that do not exist: [core.h](https://github.com/ethz-adrl/control-toolbox/blob/v3.0.2/ct_core/include/ct/core/core.h#L21)

For the time being, version 3.0.1 works for me. When you can, please look into this.
FYI: On my system, I am running Ubuntu 16.04 with ROS Kinetic, and I installed cppad separately.

markusgft, 2020-01-08:
Dear @tsender, thanks a lot for reporting your issue, this is valuable information to us.
In order to reproduce the problem, would you mind letting us know 
 * what version of CppAD you used in your custom install
 * how (and possibly where) you installed it?

Generally, in any valid recent CppAD release and installation, the example folder https://github.com/coin-or/CppAD/tree/master/include/cppad/example should be present and the code should build -- in that sense I am very curious to learn what version you installed and how you are using it. Once we have narrowed down the problem, I'm happy to provide a fix!

tsender, 2020-01-09:
I do not know how to find the exact version of CppAD that I have, but I can tell you it is either 20190200.3 or .4 (based on the dates of each release and when I likely installed it).

I used the regular cmake process (git clone, cd build, cmake, sudo make install) and it is installed in /usr/include/cppad/ on my computer.

I ran catkin build again and here is the situation:

- ct_core and ct_optcon build successfully on their own
- In one of my catkin packages, one header file contains the line: #include <ct/optcon/optcon.h>

This include in my code is what is triggering the error. I also checked that my installed version of cppad contains the cppad/example/eigen_mat_inv.hpp" file, as this file exists in: /usr/include/cppad/example/atomic_two/eigen_mat_inv.hpp

I even tried to add the #Include <cppad/example/eigen_mat_inv.hpp> in one of my own files, and again, the compiler says it can't find the file. Any ideas why it wouldn't be able to find that one file? Thanks.

markusgft, 2020-01-09:
Thanks for the detailed report. I can reproduce the issue. A patch will follow as soon as my time allows!

markusgft, 2020-01-09:
@tsender please check out the above mentioned branch and test if this works for you. If yes, I will merge.

tsender, 2020-01-09:
@markusgft I checkout out the bugfix branch, ran a catkin clean on my system and everything compiled successfully (aside from warnings about a few dependencies I don't have). I also tested the LQR solver in my gazebo simulation for verification and it ran without errors. So, you should be good to merge the change.

Thank you very much for the quick turn around!

FYI: I have been using 2019 CppAD releases in my own repo without ever having to include "cppad/example/..." in any header, and I have never had a problem. Removing those includes should not cause any harm.

closed 2020-01-10 by markusgft