ethz-adrl/control-toolbox #71: Build issure(v3.0.1)

JunLi-HIT, 2020-01-12:
I am on Ubuntu18.04 with gcc 7.4.0. before installing the control-toolbox, I have installed the kindr by Building with cmake.  I used the the scripts install_cppadcg.sh , install_hpipm.sh, and install_deps.sh to install all the dependencies.  Then, I run the build script:
    cd your_workspace/control-toolbox/ct
    chmod 775 build_ct.sh
    chmod 775 build_doc.sh
    ./build_ct.sh -DCMAKE_BUILD_TYPE=Release
I'm getting a compilation error now:
In file included from /usr/local/include/ct/core/internal/autodiff/CGHelpers.h:9:0,
                 from /usr/local/include/ct/core/systems/linearizer/DynamicsLinearizerADCG.h:8,
                 from /usr/local/include/ct/core/Systems:19,
                 from /usr/local/include/ct/core/core.h:32,
                 from /home/junli/software/control-toolbox/ct_models/include/ct/models/InvertedPendulum/codegen/InvertedPendulumActDynLinearizedForward.h:8,
                 from /home/junli/software/control-toolbox/ct_models/include/ct/models/InvertedPendulum/codegen/InvertedPendulumActDynLinearizedForward.cpp:6:
/usr/local/include/ct/external/cppad/local/jacobian.hpp:200:45: error: invalid use of incomplete type ‘class CppAD::ADFun<Base>’
 Vector ADFun<Base>::Jacobian(const Vector &x)
                                             ^
compilation terminated due to -Wfatal-errors.
CMakeFiles/InvertedPendulumActDynLinearizedForward.dir/build.make:62: recipe for target 'CMakeFiles/InvertedPendulumActDynLinearizedForward.dir/include/ct/models/InvertedPendulum/codegen/InvertedPendulumActDynLinearizedForward.cpp.o' failed
make[2]: *** [CMakeFiles/InvertedPendulumActDynLinearizedForward.dir/include/ct/models/InvertedPendulum/codegen/InvertedPendulumActDynLinearizedForward.cpp.o] Error 1
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/InvertedPendulumActDynLinearizedForward.dir/all' failed
make[1]: *** [CMakeFiles/InvertedPendulumActDynLinearizedForward.dir/all] Error 2
In file included from /usr/local/include/ct/core/internal/autodiff/CGHelpers.h:9:0,
                 from /usr/local/include/ct/core/systems/linearizer/DynamicsLinearizerADCG.h:8,
                 from /usr/local/include/ct/core/Systems:19,
                 from /usr/local/include/ct/core/core.h:32,
                 from /home/junli/software/control-toolbox/ct_models/include/ct/models/HyA/codegen/HyALinearizedForward.h:8,
                 from /home/junli/software/control-toolbox/ct_models/include/ct/models/HyA/codegen/HyALinearizedForward.cpp:6:
/usr/local/include/ct/external/cppad/local/jacobian.hpp:200:45: error: invalid use of incomplete type ‘class CppAD::ADFun<Base>’
 Vector ADFun<Base>::Jacobian(const Vector &x)
                                             ^
compilation terminated due to -Wfatal-errors.
CMakeFiles/HyALinearizedForward.dir/build.make:62: recipe for target 'CMakeFiles/HyALinearizedForward.dir/include/ct/models/HyA/codegen/HyALinearizedForward.cpp.o' failed
make[2]: *** [CMakeFiles/HyALinearizedForward.dir/include/ct/models/HyA/codegen/HyALinearizedForward.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/HyALinearizedForward.dir/all' failed
make[1]: *** [CMakeFiles/HyALinearizedForward.dir/all] Error 2
In file included from /usr/local/include/ct/core/internal/autodiff/CGHelpers.h:9:0,
                 from /usr/local/include/ct/core/systems/linearizer/DynamicsLinearizerADCG.h:8,
                 from /usr/local/include/ct/core/Systems:19,
                 from /usr/local/include/ct/core/core.h:32,
                 from /home/junli/software/control-toolbox/ct_models/src/InvertedPendulum/codegen/InvertedPendulumWithActuatorCodeGen.cpp:10:
/usr/local/include/ct/external/cppad/local/jacobian.hpp:200:45: error: invalid use of incomplete type ‘class CppAD::ADFun<Base>’
 Vector ADFun<Base>::Jacobian(const Vector &x)
                                             ^
compilation terminated due to -Wfatal-errors.
CMakeFiles/InvertedPendulumWithActuatorCodeGen.dir/build.make:62: recipe for target 'CMakeFiles/InvertedPendulumWithActuatorCodeGen.dir/src/InvertedPendulum/codegen/InvertedPendulumWithActuatorCodeGen.cpp.o' failed
make[2]: *** [CMakeFiles/InvertedPendulumWithActuatorCodeGen.dir/src/InvertedPendulum/codegen/InvertedPendulumWithActuatorCodeGen.cpp.o] Error 1
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/InvertedPendulumWithActuatorCodeGen.dir/all' failed
make[1]: *** [CMakeFiles/InvertedPendulumWithActuatorCodeGen.dir/all] Error 2
In file included from /usr/local/include/ct/core/internal/autodiff/CGHelpers.h:9:0,
                 from /usr/local/include/ct/core/systems/linearizer/DynamicsLinearizerADCG.h:8,
                 from /usr/local/include/ct/core/Systems:19,
                 from /usr/local/include/ct/core/core.h:32,
                 from /home/junli/software/control-toolbox/ct_models/src/HyA/codegen/HyALinearizationCodeGen.cpp:7:
/usr/local/include/ct/external/cppad/local/jacobian.hpp:200:45: error: invalid use of incomplete type ‘class CppAD::ADFun<Base>’
 Vector ADFun<Base>::Jacobian(const Vector &x)
                                             ^
compilation terminated due to -Wfatal-errors.
CMakeFiles/HyALinearizationCodegen.dir/build.make:62: recipe for target 'CMakeFiles/HyALinearizationCodegen.dir/src/HyA/codegen/HyALinearizationCodeGen.cpp.o' failed
make[2]: *** [CMakeFiles/HyALinearizationCodegen.dir/src/HyA/codegen/HyALinearizationCodeGen.cpp.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/HyALinearizationCodegen.dir/all' failed
make[1]: *** [CMakeFiles/HyALinearizationCodegen.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
In file included from /usr/local/include/ct/core/internal/autodiff/CGHelpers.h:9:0,
                 from /usr/local/include/ct/core/systems/linearizer/DynamicsLinearizerADCG.h:8,
                 from /usr/local/include/ct/core/Systems:19,
                 from /usr/local/include/ct/core/core.h:32,
                 from /home/junli/software/control-toolbox/ct_models/include/ct/models/HyA/codegen/HyALinearizedForward.h:8,
                 from /home/junli/software/control-toolbox/ct_models/include/ct/models/HyA/codegen/HyALinearizedForward.cpp:6:
/usr/local/include/ct/external/cppad/local/jacobian.hpp:200:45: error: invalid use of incomplete type ‘class CppAD::ADFun<Base>’
 Vector ADFun<Base>::Jacobian(const Vector &x)
                                             ^
compilation terminated due to -Wfatal-errors.
make[2]: *** [CMakeFiles/HyALinearizedForward.dir/include/ct/models/HyA/codegen/HyALinearizedForward.cpp.o] Error 1
make[1]: *** [CMakeFiles/HyALinearizedForward.dir/all] Error 2
make: *** [all] Error 2
 === completed  ct_models  === 

Could you give me some advice on how to solve them?

markusgft, 2020-01-12:
Dear @JunLi-HIT, my first guess would be the following ... did you run the scripts for installing dependencies in the correct oder? I.e. first the script "install_deps" and then the others?

JunLi-HIT, 2020-01-13:
Dear @markusgft , I have tested your guess. I installed a new Ubuntu 18.04.3 LTS system in a Desktop computer, and installed  these software again by running the following commands.
`sudo apt-get update
sudo apt-get install libeigen3-dev
sudo apt-get install libboost-all-dev
mkdir SwInstSrc
cd SwInstSrc/
git clone https://github.com/ANYbotics/kindr.git
cd kindr/
mkdir build
cd build
cmake ..
sudo make install`
Then, the kindr has been imstalled sucessfully. 

I downloaded the CT 3.0.1 manually, moved it to the worspace SwInstSrc, extracted it in the workspace, renamed it from control-toolbox-3.0.1 to control-toolbox. Then, I copied the install_cppadcg.sh and install_hpipm.sh in CT v3.0.2 to  ~/SwInstSrc/control-toolbox/ct , opened a new terminal and ran these scripts.
`cd ~/SwInstSrc/control-toolbox/ct/
sudo sh install_deps.sh
sudo sh install_hpipm.sh
sudo sh install_cppadcg.sh`
No errores appeared.
I continued to run the following commands and scripts.
`chmod 775 build_ct.sh
chmod 775 build_doc.sh
./build_ct.sh -DCMAKE_BUILD_TYPE=Release
`
Unfortunately, the same compilation errors appeared again. I believe it is not caused the execution order of these .sh files.

markusgft, 2020-01-26:
Dear @JunLi-HIT , I have just merged a few fixes to the branch v3.0.2 -- would you mind checking that one out instead and try to build?
Please note that a part of the changes is that v3.0.2 is restricted to using certain versions of CppAD and CppADCodeGen, both of them are detailed in the ./install_cppadcg.sh script.

markusgft, 2020-02-19:
close due to inactivity.

closed 2020-02-19 by markusgft