ethz-adrl/control-toolbox #78: error: ‘ADCGScalar’ is not a member of ‘ct::core’

rshum19, 2020-02-05:
Hi, 

I'm having trouble compiling my own example of the LQR Control like the example "LQR Control of a Damped Oscillator". I'm compiling my code using 

`catkin build -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true`

The error I get is 
```
In function ‘int main(int, char**)’:
/usr0/home/rshu/ct_ws/src/ct_ballbot/src/InvertedPendulumLQRExample.cpp:24:3: error: ‘ADCGScalar’ is not a member of ‘ct::core’
   ct::core::ADCGScalar g(9.81);
   ^
/usr0/home/rshu/ct_ws/src/ct_ballbot/src/InvertedPendulumLQRExample.cpp:25:3: error: ‘ADCGScalar’ is not a member of ‘ct::core’
   ct::core::ADCGScalar L(1.0);
```

below is a snippet of the code where I'm trying to define SCALAR type make my system aut-differentiable
```
ct::core::ADCGScalar g(9.81);
ct::core::ADCGScalar L(1.0);
std::shared_ptr<InvertedPendulum> invertedPendulum(new InvertedPendulum(g, L));
```

The weird thing is that the LQR Control of a Damped Oscillator compiles fine. I able to run
`rosrun ct_optcon ex_LQR` with no issues. 

I used the scripts install_deps.sh and install_cppadcg.sh to install the dependencies.

**System:** 
- Ubuntu 16.04 + ROS Kinetic
- Compiler: GCC
-  CT Version 3.0.2

Any advice would be really appreciate it thank you!

markusgft, 2020-02-05:
Hi @rshum19 , thanks for reporting!
Can you quickly verify that the ct_core unit tests pass on your system, e.g. try to run
`$ catkin build ct_core -DBUILD_TESTS=true -DCMAKE_BUILD_TYPE=Release`
afterwards you should be able to run the code-gen unit tests. `rosrun ct_core CodegenTests`
Also, please let me know if you are on branch v3.0.2 or if you are on tag 3.0.2 (they are slightly different as of now).
Please let me know what your findings are!

tsaoyu, 2020-02-05:
Having the same issue and a similar system configuration. Error message `‘ADCGScalar’ is not a member of ‘ct::core’` while trying to compile the example in a separated catkin package. All unit tests are passed on git branch v3.0.2:
```
[----------] Global test environment tear-down
[==========] 13 tests from 6 test suites ran. (18745 ms total)
[  PASSED  ] 13 tests.
```
Also did some experiment to manually turn on `CPPAD` and `CPPADCG` flag in CMakeLists. `‘ADCGScalar’` issue goes away but complains  `note:   no known conversion for argument 1 from ‘const CppAD::cg::CG<double>’ to ‘double’`.

markusgft, 2020-02-05:
Dear @tsaoyu 
Thanks for reporting. In order to clarify the issue, I wrote a quick test package, which is public and can be found [here](https://github.com/markusgft/test_package). It contains a slightly modified version of the LQR example code.
This package compiles fine on my machine using 
`catkin build test_package`.
Can you verify if this test package works or fails on your machine

rshum19, 2020-02-05:
Dear @markusgft 
Thank you for your prompt response. Similar to @tsaoyu I can also report that all the units test passed on the git branch v3.0.2

```
[----------] Global test environment tear-down
[==========] 13 tests from 6 test cases ran. (21862 ms total)
[  PASSED  ] 13 tests.
```

With the test_package I have not issues compiling. However, when I run the example 
```
rosrun test_package testLqr
```
I get the following output

```
A: 
    0     1
-2500  -100

B: 
0
1

Q: 
1 0
0 1

R: 
1

terminate called after throwing an instance of 'std::runtime_error'
  what():  solveSchurDirect() in CARE can only be used if the lapack library is installed on your system.
Aborted (core dumped)
```

It seems that the issue with   `ct::core::ADCGScalar` is not a problem with the test package. However I don't know why I get the issue with the lapack library since it is installed in my sytem  

```
rshu@gs16678:~/ct_ws$ dpkg -l liblapack-dev
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                   Version          Architecture     Description
+++-======================-================-================-==================================================
ii  liblapack-dev          3.6.0-2ubuntu2   amd64            Library of linear algebra routines 3 - static vers
```

Are there any additional steps besides `sudo apt-get install liblapack-dev` or compile flags required?

Thank you!

markusgft, 2020-02-05:
Okay .. this is getting more interesting by the minute. It might be that the find-script needs some more tweaks and tricks.
Can you confirm that lapack is not found during building? When you build with `catkin build -- verbose`, you should see some output like this when it builds ct_optcon
[image]
What is your output for lapack? 
Now, back to answer the original problem ... what was the difference in your cmakelist.txt, that made it not find the AD-Scalar type? Did you find that out?

rshum19, 2020-02-05:
Answering the original question the difference in CMakeList.txt file was a mistake in  ` target_link_libraries`, I was linking against `ct_rbd`  

**Original** 
```
add_executable( InvertedPendulumLQR src/InvertedPendulumLQR.cpp)
target_link_libraries(InvertedPendulumLQR  ct_rbd)
```

**Solved**
```
add_executable( InvertedPendulumLQR src/InvertedPendulumLQR.cpp)
target_link_libraries(InvertedPendulumLQR  ${catkin_LIBRARIES})
```

Back to the **lapack** issue it does show that lapack was found
[image]

tsaoyu, 2020-02-06:
My test result on `test_package` also shows lapack was missing on my system. Rebuild `ct_optcon` after `catkin clean` is successful with IPOPT support.  
```
-- A library with BLAS API found.
-- Found LAPACK library
-- No ENV variable for blasfeo found.
-- No ENV variable for HPIPM found.
-- Found IPOPT - building with IPOPT support
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components:  doxygen
```

markusgft, 2020-02-06:
So, @tsaoyu it is working for you, including lapack support?
And @rshum19 it is not working for you? Did you attempt a `catkin clean`?

tsaoyu, 2020-02-06:
@markusgft Still the same issue while trying to run the test package.
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  solveSchurDirect() in CARE can only be used if the lapack library is installed on your system.
Aborted (core dumped)
```
The thing is also strange that the example `ex_Nlp_2D` (require the IPOPT) runs well.

rshum19, 2020-02-06:
@markusgft 

I tried  `catkin clean` but the error in the test package still persist. However, I tried running the LQR example in ct_optcon and that had no issues

```
rosrun ct_optcon ex_LQR
```
Maybe there is something missing in the CMakeList.txt of the test_package?

markusgft, 2020-02-07:
The most likely cause for the ssue is that for some reason, the CT_USE_LAPACK compile flag gets not exported properly in your cases (on my system, it does -- which is strange). I will figure a proper fix as soon as my time allows.
In the meantime, can you simply try adding the compile definition manually to your package, e.g. something along the lines of `add_definition(CT_USE_LAPACK)`

tsaoyu, 2020-02-07:
@markusgft  add compile flag `add_definitions(-DCT_USE_LAPACK)` solve the issue. 

But the other packages discovery is still an issue. For example, the `CPPAD` flag seems broken that everytime when I call `ct::core::internal::TraitSelector<SCALAR>::Trait::cos()` the error
```
error: ‘TraitSelector’ is not a member of ‘ct::core::internal’
```

markusgft, 2020-02-07:
Dear @tsaoyu @rshum19 
I finally understood the issue. The problem is that 
`target_link_libraries( ... ${catkin_LIBRARIES}) `
does not perform a proper import of the compile flags of the catkin targets. In order to properly import the compile definitions, you are explicitly required to link against ct_optcon ...
`target_link_libraries( ... ${catkin_LIBRARIES} ct_optcon) `
..should solve the first problem.
The problem with the trait definition is, that the TraitSelector does not live in the internal namespace. The correct expression is `ct::core::tpl::TraitSelector<SCALAR>::Trait::cos()`.
I believe that you used that namespace because of an error in the documentation. I will fix that as soon as possible and push a new version of the docu!

rshum19, 2020-02-07:
Dear @markusgft 

Linking  against ct_optcon does solve the issue for the test_package. The interesting thing is that on my own package linking only to ${catkin_LIBRARIES} is sufficient. Nonetheless, I think you have helped me with my issues. I appreciate your prompt response and help!

tsaoyu, 2020-02-10:
Thank you for the help @markusgft! Now I am able to template my own model using traits.

closed 2020-02-11 by markusgft