ethz-adrl/control-toolbox #80: Issues trying to initialize a ct::core::ControlSimulator

rshum19, 2020-02-10:
Hello,

I'm struggling to understand how to build a ct::core::ControlSimulator instance. This are my lines of code

```
const size_t state_dim = ct::core::SecondOrderSystem::STATE_DIM;      // = 2
const size_t control_dim =  ct::core::SecondOrderSystem::CONTROL_DIM;  // = 1

ct::core::StateVector<state_dim> x0;
x0.setZero();

double sim_dt = 0.005;
double w_n = 50;
std::shared_ptr<ct::core::SecondOrderSystem> oscillator(new ct::core::SecondOrderSystem(w_n));
 
ct::core::ControlSimulator<ct::core::SecondOrderSystem> mpc_sim(sim_dt, sim_dt, x0, oscillator, true);
```

However, when I try to compile I get the following error.

```
/usr0/home/rshu/ct_ws/devel/include/ct/core/simulation/ControlSimulator.h:37:54: error: no type named ‘SCALAR’ in ‘class ct::core::tpl::SecondOrderSystem<double>’
     using SCALAR = typename CONTROLLED_SYSTEM::SCALAR;
```

could you please help understand what I'm doing wrong when initializing a Control Simulator.

Thank you,
Roberto

markusgft, 2020-02-11:
Dear @rshum19 
The bug above is coming from the fact that the system that you pass to the control simulator must define a typedef for the SCALAR type. That means your custom system must provide that typedef.
Please be warned about the following: The control simulator is a legacy class which has not been used for quite a while. I would recommend you to resort to the standard integrator implementation instead (see also masspoint integration example).

rshum19, 2020-02-11:
Dear @markusgft 

Thank you for the advice, if it a legacy class then I will refrain from using and resort to standard integrator implementation.

closed 2020-02-11 by rshum19