ethz-adrl/control-toolbox #81: Trouble creating a ct::core::System from ct::rbd::FixBaseFDSystem

rshum19, 2020-02-11:
Hello, 

I'v created  model of my robot using RobCoGen and I'm now trying to simulate/control it. However, I'm having trouble creating an instance of the system ct::core::System from ct::rbd::FixBaseFDSystem. When copiling my code I get the folllowing error:

```
In file included from /usr0/home/rshu/ct_ws/src/ct_ballbot/include/ct_ballbot/Ballbot2D.h:32:0, from /usr0/home/rshu/ct_ws/src/ct_ballbot/src/ballbotFDExample.cpp:3:
/usr0/home/rshu/ct_ws/devel/include/ct/rbd/robot/robcogen/robcogenHelpers.h: In instantiation of ‘static Eigen::Matrix<SCALAR, 6, ct::rbd::Ballbot2D::Utils<SCALAR>::NJOINTS> ct::rbd::Ballbot2D::Utils<SCALAR>::getJacobianBaseEEbyId(JACS&, std::size_t, const Eigen::Matrix<SCALAR, ct::rbd::Ballbot2D::Utils<SCALAR>::NJOINTS, 1>&) [with JACS = iit::ct_Ballbot2D::tpl::Jacobians<iit::rbd::DoubleTrait>; SCALAR = double; std::size_t = long unsigned int]’:
/usr0/home/rshu/ct_ws/devel/include/ct/rbd/robot/RobCoGenContainer.h:149:44:   required from ‘ct::rbd::RobCoGenContainer<RBDTrait, LinkDataMapT, U>::Jacobian ct::rbd::RobCoGenContainer<RBDTrait, LinkDataMapT, U>::getJacobianBaseEEbyId(std::size_t, const typename ct::rbd::JointState<ct::rbd::RobCoGenContainer<RBDTrait, LinkDataMapT, U>::NJOINTS, typename RBDTrait::S>::Position&) [with RBDTrait = iit::ct_Ballbot2D::tpl::Traits<double>; LinkDataMapT = iit::ct_Ballbot2D::LinkDataMap; U = ct::rbd::Ballbot2D::Utils<double>; ct::rbd::RobCoGenContainer<RBDTrait, LinkDataMapT, U>::Jacobian = Eigen::Matrix<double, 6, 1>; typename RBDTrait::S = double; std::size_t = long unsigned int; typename ct::rbd::JointState<ct::rbd::RobCoGenContainer<RBDTrait, LinkDataMapT, U>::NJOINTS, typename RBDTrait::S>::Position = Eigen::Matrix<double, 1, 1, 0, 1, 1>]’
/usr0/home/rshu/ct_ws/devel/include/ct/rbd/robot/jacobian/ConstraintJacobian.h:148:49:   required from ‘void ct::rbd::tpl::ConstraintJacobian<Kinematics, OUTPUTS, NJOINTS, SCALAR>::getJacobianOrigin(const RBDState_t&, ct::rbd::tpl::ConstraintJacobian<Kinematics, OUTPUTS, NJOINTS, SCALAR>::jacobian_t&) [with Kinematics = ct::rbd::Kinematics<ct::rbd::RobCoGenContainer<iit::ct_Ballbot2D::tpl::Traits<double>, iit::ct_Ballbot2D::LinkDataMap, ct::rbd::Ballbot2D::Utils<double> >, 1ul>; long unsigned int OUTPUTS = 3ul; long unsigned int NJOINTS = 1ul; SCALAR = double; ct::rbd::tpl::ConstraintJacobian<Kinematics, OUTPUTS, NJOINTS, SCALAR>::RBDState_t = ct::rbd::RBDState<1ul, double>; ct::rbd::tpl::ConstraintJacobian<Kinematics, OUTPUTS, NJOINTS, SCALAR>::jacobian_t = Eigen::Matrix<double, 3, 7, 0, 3, 7>]’
/usr0/home/rshu/ct_ws/src/ct_ballbot/src/ballbotFDExample.cpp:17:1:   required from here
/usr0/home/rshu/ct_ws/devel/include/ct/rbd/robot/robcogen/robcogenHelpers.h:613:82: error: ‘class iit::ct_Ballbot2D::tpl::Jacobians<iit::rbd::DoubleTrait>’ has no member named ‘base_J_fr_ee’
                 CT_RBD_JACOBIAN_GET_BLOCK(CT_EE0_FIRST_JOINT, CT_EE0_LAST_JOINT) =
                                                                                  ^
compilation terminated due to -Wfatal-errors.
```

If I look in the generated files from RobCoGen in the 'jacobians.h' file I can see that there is a class 
```
class Type_fr_base_J_fr_ee 
```

Here is a snippet of my code:
```
#include <ct/core/core.h>
#include <ct/rbd/rbd.h>
#include <ct_ballbot/Ballbot2D.h>

int main(int argc, char* argv[])
{
  using BallbotSystem = ct::rbd::FixBaseFDSystem<ct::rbd::Ballbot2D::Dynamics>;

  // obtain the state dimension
  const size_t STATE_DIM = BallbotSystem::STATE_DIM;
  std::cout << "Number of states: " << STATE_DIM << std::endl;

  // create an instance of the system
  std::shared_ptr<ct::core::System<STATE_DIM>> dynamics(new BallbotSystem);
}
```

I'm guessing that I'm doing something wrong when creating my rbd model. I followed the tutorial in the documentations. Any suggestions on what I wrong?

Thank you,
Roberto

markusgft, 2020-02-11:
I suspect that the error is somewhere in the details, probably a naming issue. The above error references "base_J_fr_ee", but what you have in your class is "fr_base_J_fr_ee ". Note the difference, there is an additional "fr". You might want to carefully check which frames are defined in your kindsl and dtdsl files!

rshum19, 2020-02-12:
Dear @markusgft 

I'm not sure what my bug was, but I rewrote all my .kindsl and .dtdsl files and fixed the problems I was having. I appreciate all your help!

closed 2020-02-12 by rshum19