ethz-adrl/control-toolbox #86: invalid covariant return type of LTISystem

yangcyself, 2020-02-23:
I'm getting `invalid covariant return type` compiler error when builidng a LTISystem:

```cpp
std::shared_ptr<ct::core::ControlledSystem<state_dim, control_dim> > Dynamics(
        new ct::core::LTISystem<state_dim, control_dim>(A, B));
```

```txt
error: invalid covariant return type for ‘const Eigen::Matrix<double, STATE_DIM, STATE_DIM>& ct::core::LTISystem<STATE_DIM, CONTROL_DIM>::getDerivativeState(const ct::core::StateVector<STATE_DIM>&, const ct::core::ControlVector<CONTROL_DIM>&, double)
```

I think the problem comes from the incoherency with StateMatrix StateVector ControlMatrix ControlVector types, and maybe solved as suggested in #39

markusgft, 2020-02-29:
Dear @yangcyself 
thanks for reporting, I will prioritize this bug in the next release! 
Best

markusgft, 2020-03-29:
Will be fixed v3.0.3, see also commit 5f73b9e on the devel branch.

closed 2020-03-29 by markusgft