ethz-adrl/control-toolbox #87: Trouble finding LAPACK

filipodu982, 2020-02-28:
Hello,
I am trying to use LQR in my project and I run into a problem:
"what():  solveSchurDirect() in CARE can only be used if the lapack library is installed on your system."
The issue is that I have Ubuntu and I tried installing lapack and blas in every way possible. I did sudo apt-get install libblas-dev liblapack-dev, I tried building the library myself and putting it into /usr/local/lib/ and even /usr/local/include, but none of those methods work. What should I do?

markusgft, 2020-02-29:
Dear @filipodu982 
I believe that all you need to do is explicitly link your target against ct_opcton (or ct_rbd, ct_models) see also the discussion in #78 
Could you give that a try?

filipodu982, 2020-02-29:
> Dear @filipodu982
> I believe that all you need to do is explicitly link your target against ct_opcton (or ct_rbd, ct_models) see also the discussion in #78
> Could you give that a try?

I added a line add_definitions(-DCT_USE_LAPACK) in CMakeLists and everything works like a charm. Thank you so much.

closed 2020-03-29 by markusgft