ethz-adrl/control-toolbox #98: Solving Controllability Gramian

kvmanohar22, 2020-03-15:
Hi,

I am using the library in solving optimal control problems as part of my optimal controls course. 

Although I am more interested in solving the problems from ground up, I am faced with few challenges. For eg: in solving a min. energy problem for a linear system using Hamiltonian formulation requires computing the Controllability gramian. I haven't come across on how to solve this matrix integral equation in this library.

Any help would be greatly appreciated!

markusgft, 2020-03-15:
Dear @kvmanohar22 
The Control Toolbox does currently not have default implementations of the controllability and observability Gramians.
So far, the implementation of the LTISystem class holds elementary checks for controllability and observability based on the Kalman criterions, see [here](https://github.com/ethz-adrl/control-toolbox/blob/af3ea074c8d067393eeda70e0dc016967be70ddc/ct_core/include/ct/core/systems/continuous_time/linear/LTISystem.h#L139)
The good news is: the control toolbox does have all the tools available for quickly doing your own implementation of the gramian matrix integration. 
I would recommend you to fix the time discretization, then use the [System Discretizer](https://github.com/ethz-adrl/control-toolbox/blob/af3ea074c8d067393eeda70e0dc016967be70ddc/ct_core/include/ct/core/systems/discrete_time/SystemDiscretizer.h#L44) to compute discrete-time matrices A and B, and then simply iterate over the discrete-time version of the controllability gramian matrix until you are happy with the accuracy.

kvmanohar22, 2020-03-15:
Thanks a lot @markusgft. Thought of adding this feature into the library. #99

kvmanohar22, 2020-03-17:
I have a small doubt, to solve for finite dimensional controllability gramian `W_{t_{f}}`, I am not quite sure how we would specify the system. In this case, the system would be the following

`dW/dt = f(t) where f(t) = e^{A(t_f - t)}BB^Te^{A^T(t_f - t)}.`

where on the RHS we have a matrix. I am not quite sure how we would specify such a system in the `System Discretizer`.

markusgft, 2020-03-18:
Dear @kvmanohar22 ,
well, I have to apologize, you are absolutely right. I pointed you to the wrong class. In fact, what you need is the `SensitivityApproximator`, see [here](https://github.com/ethz-adrl/control-toolbox/blob/af3ea074c8d067393eeda70e0dc016967be70ddc/ct_core/include/ct/core/integration/sensitivity/SensitivityApproximation.h#L32).
It allows you to set the continous-time system and the approximation type, e.g. Euler or Tustin. You can also change the time discretization easily.
Using this class, you should be able to implement your feature. Let me know if you have any further questions!

kvmanohar22, 2020-03-18:
No problem at all!!
I will have a look at that and get back to you!

markusgft, 2020-06-22:
Dear @kvmanohar22 
I was wondering if there are any further developments that could be merged back into the main repo? 
If yes, please let me know! Otherwise, I will close this issue.

kvmanohar22, 2020-06-22:
Hi @markusgft 
Yes, I think this issue can be closed. I haven't had time to get back to this.

closed 2020-06-23 by markusgft