ethz-adrl/control-toolbox #101: HPIPMInterface Build Errors

acxz, 2020-03-18:
When compiling with HPIPM enabled I receive the following errors:
```
/usr/include/ct/optcon/solver/lqp/HPIPMInterface-impl.hpp: In member function ‘void ct::optcon::HPIPMInterface<STATE_DIM, CONTROL_DIM>::computeLrInvArray()’:
/usr/include/ct/optcon/solver/lqp/HPIPMInterface-impl.hpp:222:11: error: ‘::d_ocp_qp_ipm_get_ric_Lr’ has not been declared; did you mean ‘d_ocp_qp_ipm_get_iter’?
  222 |         ::d_ocp_qp_ipm_get_ric_Lr(i, &workspace_, Lr.data());
      |           ^~~~~~~~~~~~~~~~~~~~~~~
      |           d_ocp_qp_ipm_get_iter
/usr/include/ct/optcon/solver/lqp/HPIPMInterface-impl.hpp: In member function ‘virtual void ct::optcon::HPIPMInterface<STATE_DIM, CONTROL_DIM>::computeFeedbackMatrices()’:
/usr/include/ct/optcon/solver/lqp/HPIPMInterface-impl.hpp:243:11: error: ‘::d_ocp_qp_ipm_get_ric_Ls’ has not been declared; did you mean ‘d_ocp_qp_ipm_get_iter’?
  243 |         ::d_ocp_qp_ipm_get_ric_Ls(i, &workspace_, Ls.data());
      |           ^~~~~~~~~~~~~~~~~~~~~~~
      |           d_ocp_qp_ipm_get_iter
/usr/include/ct/optcon/solver/lqp/HPIPMInterface-impl.hpp:250:7: error: ‘::d_ocp_qp_ipm_get_ric_P’ has not been declared; did you mean ‘d_ocp_qp_ipm_get_iter’?
  250 |     ::d_ocp_qp_ipm_get_ric_P(1, &workspace_, S1.data());
      |       ^~~~~~~~~~~~~~~~~~~~~~
      |       d_ocp_qp_ipm_get_iter
/usr/include/ct/optcon/solver/lqp/HPIPMInterface-impl.hpp: In member function ‘virtual void ct::optcon::HPIPMInterface<STATE_DIM, CONTROL_DIM>::compute_lv()’:
/usr/include/ct/optcon/solver/lqp/HPIPMInterface-impl.hpp:275:11: error: ‘::d_ocp_qp_ipm_get_ric_lr’ has not been declared; did you mean ‘d_ocp_qp_ipm_get_iter’?
  275 |         ::d_ocp_qp_ipm_get_ric_lr(i, &workspace_, lr.data());
      |           ^~~~~~~~~~~~~~~~~~~~~~~
      |           d_ocp_qp_ipm_get_iter
```

markusgft, 2020-03-18:
Hi @acxz 
On HPIPM, we are currently not on 0.1.1 anymore, but on a specific commit, see https://github.com/ethz-adrl/control-toolbox/blob/a59278f6e5ea4f3d63aaa3c5f4d3efb60690ef65/ct/install_hpipm.sh#L17
Everything should work fine once you check out the correct commit.

closed 2020-03-18 by markusgft

acxz, 2020-03-18:
Yeah I just figured that out, thx. I installed the `hpipm-git` package on ArchLinux instead of the `hpipm` package (which is at 0.1.1) and it built just fine. Thx!