ethz-adrl/control-toolbox #102: [Request] Move away from Python2 and support plotting via Python3

acxz, 2020-03-18:
As you may know, python.org has stopped supporting python2, and even though Canonical has picked up supporting python2 for the time being, it is probably a good idea to move towards a plotting codebase that works with Python3.

markusgft, 2020-03-18:
Good point, I agree that should be tackled at some point!

closed 2020-04-13 by markusgft

acxz, 2020-04-14:
I'm gonna leave this here since I think it relates. https://pybind11.readthedocs.io/en/stable/advanced/embedding.html

TLDR we embed python interpreter and use it for plotting. Here we can get full feature parity (since we are using python directly, instead of wrapping it) compared to matplotlib-cpp. We can also use different and better plotting libraries like [plot.ly](https://plotly.com/python/) instead of matplotlib.

Here is an example repo with this: https://github.com/tttapa/Pybind11-Matplotlib-Cpp

PR #116 (2020-04-13): Move plotting over to Python 3
Resolves #102 

Tested with
Python: 3.8.2
Matplotlib: 3.2.1
GCC: 9.3.0
cython: 0.29.16

This PR allows plotting to occur with Python 3. Note this is probably breaking for python 2, I have not tested python2.

I have tested all the examples and plotting works just fine. One point to keep in mind is that the histogram case did not work in `plotTest`. Commenting that plot out and testing the others however, ensures that the rest of the script works fine.

python3 updates were taken from upstream at the https://github.com/lava/matplotlib-cpp repository. They have add [... cut]