ethz-adrl/control-toolbox #104: Couldn't find executable named ex_NLOC

nielsv98, 2020-03-18:
Maybe this is a stupid question, but I am new to CT and trying to install it on a google coral dev board (mendel linux). I followed all the steps of building the repositories for ROS users, which seemed to go as it was supposed to. However, when I try to run one of the examples 'ex_NLOC' the following error pops up: Couldn't find executable named ex_NLOC below: /home/mendel/BJ_ws/src/control-toolbox/ct_optcon/examples 
In this directory I only see C++ files, so probably this is why it can not find an executable there. But I was hoping somebody could help me out here.
I am running ROS on this google coral dev board and created a workspace where both: 'kindr' and 'control-toolbox' are cloned in the 'src' folder.

markusgft, 2020-03-18:
Hi @nielsv98 
Are you looking for the executables in the src folder? Of course, they will not be there, since they get installed in different folders. During the building process (and before installation) you will find them in the build folder (of your catkin workspace). Afterwards, they will be in the moved to the cmake installation directory ... however I do not know what that is going to be in your setup. Checking build, devel, and lib folders in your workspace is certainly going to locate them.

acxz, 2020-03-18:
Just to make sure have you followed the steps shown on the [getting started](https://github.com/ethz-adrl/control-toolbox/wiki/Quickstart#for-ros-or-catkin-users) line by line?

markusgft, 2020-03-22:
@nielsv98 has your issue been resolved?

nielsv98, 2020-03-23:
Sorry for the late reply, I removed all build files from the board and tried to follow the 'getting started' steps again. This time it did work, I am not very sure of where it went wrong last time as it seems I did the same steps. However, the issue is now resolved. Thanks for the help!

closed 2020-03-23 by markusgft