ethz-adrl/control-toolbox #109: CotinuousOptConProblem with ADCodegenLinearizer

tsaoyu, 2020-03-26:
Does it possible to construct a `CotinuousOptConProblem` from `ADCodegenLinearizer` instead of `SystemLinearizer`? As far as I can understand, the former one performs auto-diff and the later one did it numerically. https://ethz-adrl.github.io/ct/ct_doc/doc/html/execution_speed.html recommend the use of  `ADCodegenLinearizer` for optimised execution speed.

markusgft, 2020-03-27:
Yes, that is of course possible. I would recommend you to take a look at the unit tests, they show you in detail how to instantiate a ADCodgenLinearizer for both continous-time and discrete-time systems.

tsaoyu, 2020-03-28:
Could you please point out which unit test file should I look at? I have gone through most of them but struggle to find an example that instantiates `ADCodgenLinearizer` in continuous time. Thanks

tsaoyu, 2020-03-28:
I tried to create a new templated type 
```c++
template <size_t STATE_DIM, size_t CONTROL_DIM, typename SCALAR = ADCGScalar>
using ContinuousOptConProblemAD = OptConProblemBase<STATE_DIM,
    CONTROL_DIM,
    ControlledSystem<STATE_DIM, CONTROL_DIM, SCALAR>,
    LinearSystem<STATE_DIM, CONTROL_DIM, SCALAR>,
    ADCodegenLinearizer<STATE_DIM, CONTROL_DIM, SCALAR>,
    SCALAR>;
```
to use templated dynamics and ADCG in `OptConProblemBase`. The error I got was 
```g++
error: no matching function for call to ‘ct::optcon::OptConProblemBase<12, 4, 
ct::core::ControlledSystem<12, 4, CppAD::AD<CppAD::cg::CG<double> > >, 
ct::core::LinearSystem<12, 4, CppAD::AD<CppAD::cg::CG<double> > >, 
ct::core::ADCodegenLinearizer<12, 4, CppAD::AD<CppAD::cg::CG<double> > >, 
CppAD::AD<CppAD::cg::CG<double> > >::OptConProblemBase(ct::core::Time&, const 
ct::core::StateVector<12, double>&, 
std::shared_ptr<rov::tpl::ROV<CppAD::AD<CppAD::cg::CG<double> > > >&, 
std::shared_ptr<ct::optcon::CostFunctionQuadratic<12, 4> >&, 
std::shared_ptr<ct::core::ADCodegenLinearizer<12, 4> >&)’
                 timeHorizon, x_init, rovdynamic, costFunction, adLinearizer);
```
Is this the right way to go?

markusgft, 2020-03-29:
A working example of how to use the ADCodegenLinearizer can be found in [this example in ct_optcon](https://github.com/ethz-adrl/control-toolbox/blob/f8b15baf2dd1796acb56d74876e3007cc602ca9d/ct_optcon/examples/LQR.cpp#L23).
I hope it helps!

tsaoyu, 2020-03-29:
@markusgft Thanks! I have done this already. The problem I had is how to pass the adLinearizer to ContinuousOptConProblem. The ContinuousOptConProblem seems templated for SystemLinearizer which is numerical differentiation.

```c++
template <size_t STATE_DIM, size_t CONTROL_DIM, typename SCALAR = double>
using ContinuousOptConProblem = OptConProblemBase<STATE_DIM,
    CONTROL_DIM,
    core::ControlledSystem<STATE_DIM, CONTROL_DIM, SCALAR>,
    core::LinearSystem<STATE_DIM, CONTROL_DIM, SCALAR>,
    core::SystemLinearizer<STATE_DIM, CONTROL_DIM, SCALAR>,
    SCALAR>;
```

Please advice.

markusgft, 2020-03-29:
What you try above, the "new template type" is certainly not going to work. You canot template your problem onto a fully abstract base class. 
What I would advise you to do, is to take exactly your original setup, but rather than handing over a SystemLinearizer pointer, you hand over a ADCodegenLinearizer pointer to the continous-time optcon problem. Since they derive from the same base class (LinearSystem), this is guaranteed to work.

tsaoyu, 2020-03-29:
Yep, I agree on the original setup should works. To better illustrate my issue, I have created a minimal case using the example from `ct_optcon` (the one you pointed). The code and error message is here: https://gist.github.com/tsaoyu/fd084f91c0aff21ce63dcf47d40649c2
A shared pointer is now been passed onto `ct::optcon::ContinuousOptConProblem` as I did before. In theory, this should match the function call of
```c++
    OptConProblemBase(const time_t tf,
        const state_vector_t& x0,
        DynamicsPtr_t nonlinDynamics,
        CostFunctionPtr_t costFunction,
        LinearPtr_t linearSystem = nullptr);
```
This is my point of confusion. Thanks

tsaoyu, 2020-03-29:
Thanks @markusgft !
In case anyone who runs into the same issue the answer is here: https://gist.github.com/tsaoyu/fd084f91c0aff21ce63dcf47d40649c2#gistcomment-3231924
TL;DR

Create TWO instances of dynamics and use the double templated for the `ContinuousOptConProblem`.

closed 2020-03-29 by tsaoyu

awa152, 2020-12-15:
That helped a lot, thanks!