ethz-adrl/control-toolbox #118: Updating General Constraints within MPC loop

hmcm, 2020-04-20:
I am using the MPC class for a differential drive robot in order to track a moving target. In order to avoid a collision with the moving target, I want to implement an ObstacleConstraint around the target. I have added the obstacle constraint as a general constraint (ConstraintContainerAnalytical) to the initial ContinuousOptconProblem. However, I have to update the general constraint in each MPC iteration since the position of the obstacle is changing constantly. What would be the best way to do so?

markusgft, 2020-04-22:
Hi @hmcm 
What I would recommend you to do is to use one of the non-const getters for accessing the constraint containers in the NLOptconSolver interface class, see [here](https://github.com/ethz-adrl/control-toolbox/blob/7412e0a03086ca37b6d68bbe41b88f1fdf76efa8/ct_optcon/include/ct/optcon/solver/NLOptConSolver.hpp#L237)
It will return you a std::vector of constraints, over which you can iterate and change the "setpoint" of each constraint. Please note that if you are using a multi-threading setup, say with N workers, the size of the std::vector will be N+1, so you need to change the constraint setpoint in N+1 instances.
Hope this helps!

markusgft, 2020-04-24:
I assume this has been resolved, please feel free to reopen the issue if you have further questions.

closed 2020-04-24 by markusgft

hmcm, 2020-04-24:
Hi @markusgft, 
thank you for this fast answer, it solved my issue!