ethz-adrl/control-toolbox #122: Convex MPC

mahumada408, 2020-04-27:
Hi,

I wanted to know if you have any thoughts on where I should get started to solve convex MPC problems with the control toolbox. For example, would you be able to use this architecture to solve the MPC problem in the MIT Cheetah (and mini Cheetah), and use Gazebo as the simulation platform? I have a gazebo sim of a small quadruped ready to go, and I'd like to control it with an MPC. If you could give me any tips, I would highly appreciate it.

I'd love to give this a shot!

Cheetah Paper
https://doi.org/10.1109/IROS.2018.8594448

Thank!

markusgft, 2020-04-27:
Dear @mahumada408 
The CT does unfortunately not come with a default gazebo integration, but it is certainly a tool which can help you in solving the mentioned MPC problems.
For communicating with gazebo, you could for instance use ROS. If you are interested in seeing CT examples with ROS integration, including a motion optimization task for a quadruped, I can recommend you checking out the examples in the accompanying repository [ct_ros](https://github.com/ethz-adrl/ct_ros).
Best!

mahumada408, 2020-04-27:
Great! I'll take a look at ct_ros. 
Are there any examples on how to solve a convex MPC with CT?

markusgft, 2020-04-28:
Well, basically all the linear-quadratic examples and unit tests are convex problems, which may serve as examples for you!

erwincoumans, 2020-04-30:
You may also get some inspiration from their repository, both control and simulator of the MIT Cheetah is open source here: https://github.com/mit-biomimetics/Cheetah-Software
(search for LQR and QP)

mahumada408, 2020-05-01:
@markusgft thanks for pointing out the linear quadradic examples. I realize that most of those are LQR problems with no constraints. How could I set the problem up with constraints? At that point, I assume I would need to use IPOPT, right?

@erwincoumans thanks, I didn't realize that the control code for Cheetah was open source. I'll definitely take a look and try to implement it with this toolbox in ROS!

markusgft, 2020-05-01:
[Here ](https://github.com/ethz-adrl/control-toolbox/blob/4fd7c90d1c9ef329b2cc596be45e76621a16e8ed/ct_optcon/examples/NLOC_boxConstrained.cpp#L17) is an example with box constraints, in the same folder there is an example which uses general inequality constraints. 
The solver which is employed for the constrained LQ problems is HPIPM, which you can install using the convenience bash script in the /ct folder.

closed 2020-05-08 by markusgft