ethz-adrl/control-toolbox #126: Compile problem

crankler, 2020-05-03:
Hi, I'm strange to "catkin_build" so I follow the quickstart:https://github.com/ethz-adrl/control-toolbox/wiki/Quickstart

When I try to compile this package,  I use the command: 
`catkin build -DCMAKE_BUILD_TYPE=Release
Then I inout the command and error show：
`catkin build -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true`

> ../control-toolbox/ct_core/examples/src/plot/plotTest.cpp:36:15: error: ‘ct::core::plot’ has not been declared
     ct::core::plot::plot(v);
               ^~~~
compilation terminated due to -Wfatal-errors.

I'm not quite sure where is the problem for this issus, can anybody give me some advice？ 
Thank you!

markusgft, 2020-05-03:
Hi @crankler 
this was a problem on our side -- I pushed some commits to fix it. Please pull again and retry!
Best

crankler, 2020-05-03:
I have tried and it works, Thank you for your help and sharing!

closed 2020-05-04 by markusgft