ethz-adrl/control-toolbox #130: Logging to Matlab

rshum19, 2020-05-11:
Hi @markusgft 

I'm continuing to explore more the CT and all its features. I came across some the "MATLAB" flag in ct_optcon/CMakeLists.txt and the "logToMatlab" solver settings. I was wondering if there exist a capability to log to Matlab. If so, how do I actually use it. I tried setting the "logToMatlab = True" but that didn't seem to do anything. 

Thanks!

markusgft, 2020-05-12:
Dear @rshum19 
yes, there exists the possibility to log to Matlab, however this requires a separate package, which is as of today not open-source. Let me check if we can open-source that package, I will come back to this thread.
Best

markusgft, 2020-05-12:
Please check out [this](https://github.com/ethz-adrl/matlab_cpp_interface) and confirm that it works for you (or not).
Best

rshum19, 2020-05-12:
Dear @markusgft  Thank you for getting back so quick and open sourcing the matlab interface package!!

 I have compiled the matlab_cpp_interface package successfully and all the the test seem to pass from test_main.cpp and ros_test_main.cpp.

However, I'm not quite sure how to log to matlab. I have 

1.  Recompiled the **ct_opcon** package with the flag `-DMATLAB=ON`
2.  In my solver.info I have set `logToMatlab true` and `loggingPrefix "~/ct_ws/ct_Ballbot/matlab"`

I can't find any saved MAT files. 

I also tried the hya_nloc_goto example in ct_ros_nodes package, but also couldn't find any MAT files.

markusgft, 2020-05-13:
Dear @rshum19 
I have not been using the matlab interface for almost 3 years, so I can only make a quick guess for debugging. 
 * Maybe, the easiest starting point for you would be to include a simple printout at the place where you expect the logging function to be executed, and verify that it actually calls it? This may be somewhere in the NLOCBackendBase class.
 * I am not sure if the engine can resolve the path beginning with `~`, maybe try an absolute path 
 * Also, there are different levels of logging possible. Can you try to compile with `-DMATLAB_FULL_LOG=true` ?

rshum19, 2020-05-13:
@markusgft  thank you for the suggestions, I will try them and report back if I was successful in getting it working.  

I appreciate your prompt response, the support you provide for this library is one of the best features!

closed 2020-05-19 by markusgft