ethz-adrl/control-toolbox #133: Solving quadratically constrained quadratic problems

hmcm, 2020-05-19:
Hi, 
I am using the MPC class for a differential drive robot in order to track a moving target.  In order to avoid a collision with the moving target and other dynamic obstacles, I used the implemented ObstacleConstraint. However, it appears that HPIPM is not able to solve the problem with the obstacle constraint. The solver returns flag 2, "minimum step length reached". 

I have reduced the problem to a single NLOC (without MPC loop), similar to the example ex_NLOC_generalConstrained, which I have attached along with the configuration files. I hope it is executable. 
Do you have any suggestions? Or maybe an example of an implementation of the ObstacleConstraint? I have been wondering whether this issue arises due to the quadratic form of the constraint or whether it is an implementation error from my side regarding the ObstacleConstraint. 

Thanks a lot!
[NLOC_ObstacleConstraint.tar.gz](https://github.com/ethz-adrl/control-toolbox/files/4650760/NLOC_ObstacleConstraint.tar.gz)

markusgft, 2020-05-19:
Hi @hmcm 
I general I do not see any major obstacles which should prevent hpipm from solving your problem, so it is most likely some kind of configuration issue. Let's debug this step by step. 

 * First of all, can you verify that the problem is reasonably solved when you remove both box and general constraints, and only consider the differential-drive dynamics? If your cost function is well-tuned, you should end up very closely to your desired final configuration.

 * Secondly, let's leave the box constraints away for the time being and only consider the general constraints. Do you get the same error when leaving away the box constraints entirely?

hmcm, 2020-05-19:
Hi @markusgft , 
thanks for your fast reply!

1)  If I remove all constraints, the goal position is reached with a smooth trajectory. HPIPM returns that the QP was solved. Maybe as an additional point here: I have been using the MPC with the differential-drive dynamics without the obstacle constraint (but with box constraints), and it behaves as expected.

2)  If I remove the box constraints and only consider the general constraints, HPIPM returns with flag 3, "NaN in computations" (By the way, is it possible that there is a mistake in the HPIPMInterface-impl in line 201? You consider hpipm_status_==2 twice, but do not consider hpipm_status_==3, see https://github.com/ethz-adrl/control-toolbox/blob/5a9ed9c8ec4dc78bc9067e2b27563610117725cb/ct_optcon/include/ct/optcon/solver/lqp/HPIPMInterface-impl.hpp#L201). 

3)  If I consider both box constraints and general constraints and solve using ILQR, I have the same issue (flag 2, minimum step length reached). Also for 1) and 2), I get the same behaviour as with GNMS.

markusgft, 2020-05-20:
Hi @hmcm 
Thanks for pointing this out, this is clearly a bug, of course it should be hpipm_status_ == 3.

hmcm, 2020-05-20:
Hi @markusgft 

I opened the pull request for the patch. As for the feature, that sounds very promising!

Regarding my issue: When I move the obstacle to a completely different region (pos_obstacle = (-10,-10) when goal = (10,10)), I still get the same numerical problems with only the general constraints. If I add the box constraints, hpipm still returns flag 2. (both for GNMS and ILQR)

markusgft, 2020-05-21:
Hi @hmcm 
I just looked at your example and I believe I found potential problems.
Firstly -- and this is a problem on our end -- you are using a constraint container which was never really tested in combination with the NLOC solver. The obstacle constraint class was originally developped for the NLP-based solvers, and honestly, without going through a test procedure I cannot tell you right away if they work together. What I would recommend you is to quickly implement the obstacle- inequality constraint from scratch, following the example in ex_NLOC_generalConstrained.cpp.
Next, when working with HPIPM, it would make sense to reduce the dimension of the constraints to the actually numerically relevant order. The third row of your inquality constraint, which represents orientation, is redundant and the constraint Jacobian is set to zero all the time. This may lead to issues in hpipm. I would recommend you to use x- and y coordinates only in a custom implementation of the general constraint. 
If you do this, I am pretty sure it is going to work!

hmcm, 2020-05-22:
Hi @markusgft 
Thanks a lot for looking into it! I will try to implement your approach next week and see if I can get it working!

hmcm, 2020-05-25:
Hi @markusgft , 
I implemented the obstacle constraint from scratch and reduced the dimensions to x and y (implementation attached). To make it as simple as possible, I consider the obstacle as a unit circle and only provide the center coordinates as an input for the class constructor. However, I still get exactly the same problems. I have tried using the automatic differentiation as in the ex_NLOC_generalConstrained.cpp as well, but this didn't change the result either... Did I misunderstand any of your instructions? Or is this what you had in mind?
[Reimplementation_ObstacleConstraint.tar.gz](https://github.com/ethz-adrl/control-toolbox/files/4677647/Reimplementation_ObstacleConstraint.tar.gz)

markusgft, 2020-05-25:
Hi @hmcm, could you quickly try with the following settings for the nloc solver
```
ilqr
{
    nlocp_algorithm GNMS
    integrator EULER
    useSensitivityIntegrator false
    discretization Forward_euler
    timeVaryingDiscretization false
    dt 0.005
    K_sim 1
    K_shot 1
    epsilon 0
    max_iterations 100
    fixedHessianCorrection false
    recordSmallestEigenvalue false
    min_cost_improvement 1e-5
    maxDefectSum 1e-5
    meritFunctionRho 0.0
    meritFunctionRhoConstraints 0.0
    nThreads 1
    nThreadsEigen 1
    locp_solver HPIPM_SOLVER
    printSummary true
    debugPrint false 
    logToMatlab 0   
    
    line_search
    {
        type NONE
        adaptive false
        maxIterations 10
        alpha_0 1.0
        n_alpha 0.5
        debugPrint true
    }

    lqoc_solver_settings
    {
        lqoc_debug_print        true
        num_lqoc_iterations  	5
    }
}
```

hmcm, 2020-05-26:
Hi @markusgft,

I tried it and still get the "minimum step length reached" flag. Anything in particular that I should pay attention to?

markusgft, 2020-05-26:
Hi @hmcm 
I think we are at the point where we need to validate that it works in a setting without the differential constraint. Can you try formulating the dynamics as a second-order dynamic system, i.e. simply a moving point mass? Ideally, could you also put this in a repo that you can give me access to, so that I can jump in with debugging in case we need it?
Best
Markus

hmcm, 2020-05-26:
Hi @markusgft 
I will try that, but unfortunately I won't have time before next week. Thanks a lot for the help so far and I will get back to you as soon as I have reformulated the dynamics and put it into a repo!
Best
Hannah

markusgft, 2020-06-04:
Hi @hmcm, I would still be interested in following up on your issue and debugging the problem. Did you by any chance find time to look into the modified example?
Best!

hmcm, 2020-06-05:
Hi @markusgft 
I am sorry this took so long, but I had a lot going on. I tried to implement a second order system as you proposed. However, I am not sure I understood you correctly. I just implemented a point mass with four states (x, y, v_x, v_y) which is actuated with a force in x-direction and a force in y-direction. However, if I try solving the control problem for this system, the line search returns that the control is unstable. Was this the kind of system you had in mind?
I invited you as a collaborator to the repo where I uploaded everything, there I also included the constraint class. I tried figure out why the control is unstable and will get back to work on it next Thursday. Sorry for only providing you with something incomplete, but maybe it is already of some help!
Best
Hannah

markusgft, 2020-06-05:
Hi @hmcm 
Thanks for sharing your repo. I took a quick look, and indeed there is some issue that I can confirm. For instance, in my case, GNMS with constraints diverges instantly. I will try allocate some time to find and fix the problem.

markusgft, 2020-06-11:
Should be resolved by #142

closed 2020-06-12 by hmcm