ethz-adrl/control-toolbox #143: Derivatives of class LTISystem do not compile because of flawed function override

schulz0r, 2020-06-18:
Dear developer team,
I use Apple clang version 11.0.3 and cannot compile following code:
```
class Model: public ct::core::LTISystem<6, 3> {
    Model(): ct::core::LTISystem<6, 3>(Eigen::MatrixXd::Zero(6,6), Eigen::MatrixXd::Zero(6,3));
};
```
The reason is that the class `LTISystem` incorrectly overrides the virtual functions getDerivativeState() and getDerivativeControl() which originally stem from the parent class `LinearSystem`.
The code of `LTISystem.h` shows that the return type is `Eigen::Matrix`, but the compiler expects the types to be `StateMatrix` and `StateControlMatrix`. They might be the same inherently, but the compiler does not accept that.

markusgft, 2020-06-19:
Dear @schulz0r , this had been reported in #86, too. However, at that point in time, we fixed it only for the devel-branch. Since the issue was raised again for v3.0.2, I will look into this and commit a fix as soon as I find time.

markusgft, 2020-06-20:
Please check if it works for your now!

schulz0r, 2020-06-20:
My code compiles now! Thanks for the quick response!

closed 2020-06-20 by schulz0r