ethz-adrl/control-toolbox #148: Cannot find hpipm/blasfeo headers

schulz0r, 2020-06-29:
Dear developers,
I downloaded hpipm and blasfeo from their respective git repositories here on Github and built/installed them on my machine. When compiling the Control Toolbox with the optimal control addon, I encounter the problem that blasfeo_target.h cannot be found. **The problem could be solved** when changing line 66 in ct_optcon/CMakeLists.txt from:

`list(APPEND HPIPM_LIBS ${hpipm_LIBRARIES} ${blasfeo_LIBRARIES})`
to:
`list(APPEND HPIPM_LIBS hpipm blasfeo)`

CMake does not require you to add `${<name>_LIBRARIES}` and `${<name>_INCLUDE_DIRS}` to link against dependencies or include their headers. Just link the target created by `find_package()`, this will also include all headers. I believe that this way is less error prone and should be preferred.

acxz, 2020-07-10:
I agree with this. The `_LIBRARIES` and `_INCLUDE_DIRS` variables are legacy methods of handling deps. I'll see if I can reproduce the error with the latest hpipm and blasfeo and come up with a solution.

acxz, 2020-07-11:
re: python, we want to stay on a lower minimum cmake version to support legacy operating systems which is why we cannot link against the `Python:Python` target. Once we raise our min cmake version we will be sure to update that as well.

acxz, 2020-07-11:
@schulz0r if you could test out the linked PR and confirm that it resolves the issue for you that would be great. Thx for reporting.

schulz0r, 2020-07-13:
I just want to tell that I have seen your PR and I will check it out this week.

schulz0r, 2020-07-13:
I checked it out and it worked for me. Thanks a lot!

acxz, 2020-07-13:
Glad to hear!

PR #152 (2020-07-14): use blasfeo/hpipm targets
Addresses and resolves #148
closed 2020-07-14 by markusgft