ethz-adrl/control-toolbox #159: Problem Installing IPOPT

padhupradheep, 2020-08-06:
Hey,

This issue is more of a suggestion! 

I'm using ubuntu 18.04, 

Just during the process of building the control-toolbox package (with the examples), I had a problem with the Ipopt (dependency problem). Although I used their documentation to install it, I had the error of 

`/usr/include/coin/IpSmartPtr.hpp:18:4: error: #error "don't have header file for stddef"`

In order to sort the issue, I tried various options for installations, luckily the instruction given in this [link](https://gist.github.com/silgon/0892f410b025f2a82f2b) worked for me. 

Therefore, in order for future users to avoid the same problem, It would be great for you guys to have this link in your documentation, to make it a bit easier for the users. 

Also works fine with 16.04

Pradheep

acxz, 2020-08-07:
Hello @padhupradheep thx for the suggestion, but it seems like you went through a roundabout way of installing `ipopt`.
`ipopt` is available in Ubuntu 18/16 as the following package `coinor-libipopt-dev`. You can install it by just running `sudo apt install coinor-libipopt-dev`.

Let us know if this solution works for you and if it does feel free to close this issue.

padhupradheep, 2020-08-07:
Hey,

Sorry, I forgot to mention the` coinor-libipopt-dev` ! I initially installed the same and did not work. It had an error as follows: 

`/usr/include/coin/IpSmartPtr.hpp:18:4: error: #error "don't have header file for stddef"`

And then I did a bit of research about the stddef dependency and found the same error in the [installation step of Ipopt for CasAdi](https://github.com/casadi/casadi/wiki/InstallationLinux) (please scroll down a bit to find it). 

Hope this helps for someone!

acxz, 2020-08-07:
hmm interesting have you reported the issue to the [ipopt repo](https://github.com/coin-or/Ipopt), since it pertains to an error in that repo instead of this one? That issue should have been resolved in versions > 3.10.1 which ubunut 16 and upwards have.

Can you try to completely uninstall ipopt on your system and reinstall? I believe there might be something wrong on your system.

padhupradheep, 2020-08-07:
No I have not yet reported over there yet! 

Did the complete re-installation process as well! But still did not work! I tried it with both 16.04 and 18.04!

acxz, 2020-08-07:
hmm do let us know what the ipopt people say

padhupradheep, 2020-08-07:
Sure! I will close the issue, for now, I will just give the update in the same issue though! Thanks for your time!

closed 2020-08-07 by padhupradheep