ethz-adrl/control-toolbox #161: Compiler error with Python header

padhupradheep, 2020-08-07:
Hey,

I'm using Ubuntu 18.04. I had a problem with the Python header file. When I tried to build it I had the compiler error.

For the exact file details see this [PR](https://github.com/ethz-adrl/control-toolbox/pull/160/files)

Therefore I tweaked it a little bit. It did not work for python2.7 though! Had an error with strings! I do not remember the exact details of the error. 

P.S.: sudo apt install libpython3.7-dev

As pointed out in the PR, the fix I have is a hardcoded fix, I would highly recommend the developers of this package to look into it, so that it does not break the system!

acxz, 2020-08-07:
@padhupradheep thx for posting an issue, but without the error message we cannot help as we cannot reproduce the issue on our systems.

Please post the error message you received.

padhupradheep, 2020-08-07:
Updated!

acxz, 2020-08-07:
Do you have the `python3-dev` package installed?

padhupradheep, 2020-08-07:
Yep! I had!

closed 2020-10-31 by markusgft