ethz-adrl/control-toolbox #164: Case insensitive file systems

egorbarsukoff, 2020-09-11:
By default, modern Apple computers are configured with a case-insensitive file system. This creates big problems when developing with the Control toolbox on this platform, because, for example, there is a `Common` file and `common` folder in `ct_core`, which cannot exist in the same folder at the same time. This means that I cannot clone this repo on my local machine. I would very much appreciate any solution to this particular problem or proposing the way (new naming conventions) to fix with and (possible) contribute the solution.

markusgft, 2020-09-16:
Hi, we have had people encounter that issue before, and in fact someone had a workaround for it. It seems like this did not get posted on the issue tracker, I am digging for the solution in my old e-mails.

markusgft, 2020-09-16:
After doing a thorough search, I can confirm there is no quick way to fix this issue and make the CT compatible with Apple Computers. Sorry! Alternatively, I can recommend you to try a virtual machine for development!

closed 2020-09-16 by markusgft