ethz-adrl/control-toolbox #165: blasfeo_target.h：not such file

JMQKK, 2020-09-15:
I am very troubled with this problem that i can't find header file
“devel/include/ct/optcon/solver/lqp/HPIPMInterface.hpp:12:28: fatal error: blasfeo_target.h:not such file”

I try to modify the Cmakelists. txt in ct_optcon, but no effect.

markusgft, 2020-09-16:
Hi, 
would you mind providing more detail about your setup? Are you sure you only have one version of blasfeo and hpipm installed?
First and foremost, we should make sure that the examples and unit tests build on your system. After installing hpipm using the convenience script `./install_hpipm`, can you try building one of the constrained examples, for instance by `catkin build ct_optcon -DBUILD_EXAMPLES=true`?
If you are not using catkin, you will need to use the equivalent cmake command with the same compiler flag.
Please let us know what you encountered!

JMQKK, 2020-09-17:
Thank you for taking time to answer my questions
I have tried to install different versions of  blasfeo and hpipm by "cmake "
And I tried to build the example again,but  the program still couldn't find the header file "blasfeo_target.h".

markusgft, 2020-09-17:
I suppose that you might have conflicting versions of hpipm on your system. Can you please make sure that you remove all hpipm and blasfeo files/installations, and give it another try to install them using our script? You may have to check both the /opt and the /usr/local directories to eliminate all your legacy installations.

kukarkinmm, 2020-09-24:
Hi,
I've faced the same problem even after installing using your script. It seems that default installation directory of blasfeo is /opt/blasfeo instead of /usr/local/include. The solution was simply specify include directory to my target in CMake file using target_include_directories() and target_link_library() for correct linking.

JMQKK, 2020-09-25:
I copied them all to /usr/local/.Then there are not any errors.
I suoopse that "list(APPEND CMAKE_PREFIX_PATH "/opt")"didn't work.

awa152, 2020-09-29:
After installing blasfeo and hpipm with the script ./install_hpipm and building my workspace with catkin build I still get the error 
~~~
CMake Warning at /home/awahl/.../control-toolbox/ct_optcon/CMakeLists.txt:74 (message): Could not find HPIPM or BLASFEO
~~~
Copying the static library folders /blasfeo and /hpipm into /usr/local didn't worked for me. When removing the QUIET option in find_package(blasfeo QUIET) in the CMakeLists.txt of ct_optcon, I get the warning 
~~~
Warnings   << ct_optcon:cmake /home/awahl/GitLab/panda/workspaces/ws_control_toolbox/logs/ct_optcon/build.cmake.009.log              
CMake Warning at /home/awahl/GitLab/panda/workspaces/ws_control_toolbox/src/control-toolbox/ct_optcon/CMakeLists.txt:66 (find_package):
  By not providing "Findblasfeo.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "blasfeo", but
  CMake did not find one.

  Could not find a package configuration file provided by "blasfeo" with any
  of the following names:

    blasfeoConfig.cmake
    blasfeo-config.cmake

  Add the installation prefix of "blasfeo" to CMAKE_PREFIX_PATH or set
  "blasfeo_DIR" to a directory containing one of the above files.  If
  "blasfeo" provides a separate development package or SDK, be sure it has
  been installed.
~~~
As I am fairly new to the cmake build process I don't really know how to make ct_optcon find blasfeo and hpipm. The static library install from the ./install_hpipm doesn't seem to provide the .cmake files.

awa152, 2020-09-30:
Using cmake as the build system for blasfeo and hpipm solved the issue and created the necessary .cmake files for the find_package() command. 
~~~
#!/bin/bash

## get blasfeo
echo "Now installing blasfeo, using a specific commit/tag (!)"
cd /tmp
git clone https://github.com/giaf/blasfeo.git
cd /tmp/blasfeo
git checkout cc90e14 # we currently are on this commit.
mkdir build
cd build
cmake ..
sudo make install

## get hpipm
echo "Now installing hpipm, using a specific commit/tag (!)"
cd /tmp
git clone https://github.com/giaf/hpipm.git
cd /tmp/hpipm
git checkout 5d9909f # we currently are on this commit.
mkdir build
cd build
cmake ..
sudo make install
~~~
Maybe an update to the installation script like the one above would be good?

When building my workspace this resulted in an issue with building ct_models with the error 
~~~
Errors     << ct_models:make /home/awahl/GitLab/panda/workspaces/ws_control_toolbox/logs/ct_models/build.make.000.log                
In file included from /home/awahl/GitLab/panda/workspaces/ws_control_toolbox/devel/include/ct/optcon/nloc/NLOCBackendBase.hpp:16:0,
                 from /home/awahl/GitLab/panda/workspaces/ws_control_toolbox/devel/include/ct/optcon/optcon.h:25,
                 from /home/awahl/GitLab/panda/workspaces/ws_control_toolbox/devel/include/ct/rbd/rbd.h:9,
                 from /home/awahl/GitLab/panda/workspaces/ws_control_toolbox/src/control-toolbox/ct_models/src/InvertedPendulum/codegen/InvertedPendulumWithActuatorCodeGen.cpp:10:
/home/awahl/GitLab/panda/workspaces/ws_control_toolbox/devel/include/ct/optcon/solver/lqp/HPIPMInterface.hpp:12:10: fatal error: blasfeo_target.h: Datei oder Verzeichnis nicht gefunden
 #include <blasfeo_target.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
~~~

Excluding ct_models from the build with a CATKIN_IGNORE resulted in a successful build and I was able to run the examples  ex_NLOC_boxConstrained and ex_NLOC_generalConstrained. Including 
~~~
list(APPEND CMAKE_PREFIX_PATH "/opt")
find_package(blasfeo QUIET)
find_package(hpipm QUIET)
~~~
into the CMakeLists.txt of ct_models resolved the issue and the build of ct_models succeded as well.

markusgft, 2020-11-01:
Dear all, thank you for your patience. I can reproduce the issue, working on a fix as we speak.

closed 2020-11-01 by markusgft

rokasgie, 2021-02-10:
I am still receiving this error after having installed HPIPM and blasfeo with the provided script. The only solution that worked for me was to add 

```
list(APPEND CMAKE_PREFIX_PATH "/opt")
find_package(blasfeo QUIET)
find_package(hpipm QUIET)
```
to CMakeLists.txt but then when I try to use the HPIPMInterface it says that it hasn't been built:

```
error: ‘HPIPMInterface’ in namespace ‘ct::optcon’ does not name a template type
   24 |             new ct::optcon::HPIPMInterface<STATE_DIM, CONTROL_DIM>);
```
I am building using the build_ct script