ethz-adrl/control-toolbox #166: Problem with code generation in RobCoGen

awa152, 2020-09-25:
I know that this isn't an issue with the control toolbox itself, but with the generation step in RobCoGen, but I don't really know where to file an issue elsewhere.

I am trying to generate the files on Ubuntu 18.04 with openjdk-8-jre installed with the compatible RobCoGen version 0.4ad.0.
When trying to generate the C++ code everything works except the generation of the coordinate Transforms and the geometric jacobian. I tried the example.kindsl provided in the folder and a few other robots descriptions. I also reinstalled java and tried it inside a Docker Container to be sure that there is no conflict between different java versions on my laptop. In the Docker Container I tried openjdk-8-jre and openjdk-11-jre.

I get the following errors

When trying to get the C++ coordinate Transform:
~~~
Exception in thread "main" java.lang.NullPointerException
at iit.dsl.maxdsl.utils.MaximaConversionUtils.cleanMatrixRowOutput(Unknown Source)
at iit.dsl.maxdsl.utils.MaximaConversionUtils.getMatrix(Unknown Source)
at iit.dsl.coord.generator.maxima.converter.MaximaConverter.parseTransform(Unknown Source)
at iit.dsl.coord.generator.cpp.TransformsClasses.localClassesDefinitions(Unknown Source)
at iit.dsl.coord.generator.cpp.EigenFiles.implementationsFileContent(Unknown Source)
at iit.dsl.generator.cpp.kinematics.Transforms.generate(Unknown Source)
at iit.dsl.generator.cpp.Generator.generateTransforms(Unknown Source)
at iit.robcogen.KinDSLWrapper.generate(Unknown Source)
at iit.robcogen.Generator.main(Unknown Source)
~~~

When trying to get the geometric jacobian:
~~~
Exception in thread "main" java.lang.NullPointerException
at iit.dsl.maxdsl.utils.MaximaConversionUtils.cleanMatrixRowOutput(Unknown Source)
at iit.dsl.maxdsl.utils.MaximaConversionUtils.getMatrix(Unknown Source)
at iit.dsl.generator.maxima.Converter.getJacobianText(Unknown Source)
at iit.dsl.generator.cpp.kinematics.Jacobians.localClassesDefinitions(Unknown Source)
at iit.dsl.generator.cpp.kinematics.Jacobians.sourceFile(Unknown Source)
at iit.dsl.generator.cpp.Generator.generateJacobiansFiles(Unknown Source)
at iit.robcogen.KinDSLWrapper.generate(Unknown Source)
at iit.robcogen.Generator.main(Unknown Source)
~~~

Maybe someone else already had the same issue and could point me into the right direction. I tried to generate the code with the newer version 0.5.1 and it worked, but I don't know about the compatibility between the newer version 0.5.1 and the control toolbox. 

kind regards

awa152, 2020-09-25:
After writing with Marco Frigerio he showed me that this error is due to an old bug that was fixed in the newer version of RobCoGen, but not in 0.4ad.0. 
Simply removing a "/" sign from the framework.properties configuration resolved this issue, like:
generator.maxima.libs.path = ../etc/maxima-libs (do NOT use "/" at the end).

I am just writing this in case somebody else runs into the same problem when trying to use the control toolbox with RobCoGen 0.4ad.0

closed 2020-09-25 by awa152

csufangyu, 2020-10-29:
hi I get the problem that looks a liitle similar to yours, but I still cannot compile successfully in 0.5.1. If I enter 1, 4, the error is:
```
robot203@robot203:~/robcogen/building/0build$ ./robcogen.sh ../etc/example.kindsl
 0 - CTDSL	- The coordinate transforms description file .ctdsl

 1 - ALLMx	- [All the Maxima targets]
 2 - TMx	- Maxima coordinate transforms
 3 - JMx	- Maxima geometric Jacobians

 4 - ALLC++	- [All the C++ targets]
 5 - CC++	- C++ common code
 6 - TC++	- C++ coordinate Transform
 7 - JC++	- C++ geometric Jacobians
 8 - DC++	- C++ common code for dynamics
 9 - IDC++	- C++ Inverse-Dynamics implementation
10 - FDC++	- C++ Forward-Dynamics implementation
11 - JSIMC++	- C++ Joint-Space-Inertia-Matrix calculation
12 - MKC++	- CMake file for C++ code

13 - ALLMt	- [All the Matlab targets]
14 - CMt	- Matlab common code
15 - TMt	- Matlab coordinate transforms
16 - JMt	- Matlab geometric Jacobians
17 - IDMt	- Matlab Forward/Inverse-Dynamics implementation
18 - JSIMMt	- Matlab Joint-Space-Inertia-Matrix calculation

19 - SL	- [All the SL targets]
20 - SLR	- SL, robot code (includes the robot model .dyn)
21 - SLU	- SL, robot-user code

22 - RB	- [All the robot models]
23 - RB_ROS	- XML URDF format
24 - RB_Feath	- Matlab structure for Featherstone's code
25 - RB_SDFAST	- SD/FAST model descritpion

26 - Test	- Support code to launch the Octave tests for Octave/C++

27 - Reload	- Reload the input files
28 - Quit	- Exit robcogen

What would you like to generate? Please enter the integer code:
1,4
Exception in thread "main" java.util.InputMismatchException
	at java.util.Scanner.throwFor(Scanner.java:864)
	at java.util.Scanner.next(Scanner.java:1485)
	at java.util.Scanner.nextInt(Scanner.java:2117)
	at java.util.Scanner.nextInt(Scanner.java:2076)
	at iit.robcogen.Generator.main(Unknown Source)

```
If I enter 4,the error is:  
 0 - CTDSL	- The coordinate transforms description file .ctdsl

 1 - ALLMx	- [All the Maxima targets]
 2 - TMx	- Maxima coordinate transforms
 3 - JMx	- Maxima geometric Jacobians

 4 - ALLC++	- [All the C++ targets]
 5 - CC++	- C++ common code
 6 - TC++	- C++ coordinate Transform
 7 - JC++	- C++ geometric Jacobians
 8 - DC++	- C++ common code for dynamics
 9 - IDC++	- C++ Inverse-Dynamics implementation
10 - FDC++	- C++ Forward-Dynamics implementation
11 - JSIMC++	- C++ Joint-Space-Inertia-Matrix calculation
12 - MKC++	- CMake file for C++ code

13 - ALLMt	- [All the Matlab targets]
14 - CMt	- Matlab common code
15 - TMt	- Matlab coordinate transforms
16 - JMt	- Matlab geometric Jacobians
17 - IDMt	- Matlab Forward/Inverse-Dynamics implementation
18 - JSIMMt	- Matlab Joint-Space-Inertia-Matrix calculation

19 - SL	- [All the SL targets]
20 - SLR	- SL, robot code (includes the robot model .dyn)
21 - SLU	- SL, robot-user code

22 - RB	- [All the robot models]
23 - RB_ROS	- XML URDF format
24 - RB_Feath	- Matlab structure for Featherstone's code
25 - RB_SDFAST	- SD/FAST model descritpion

26 - Test	- Support code to launch the Octave tests for Octave/C++

27 - Reload	- Reload the input files
28 - Quit	- Exit robcogen

What would you like to generate? Please enter the integer code:
4
```
Exception in thread "main" java.lang.RuntimeException: Could not find Maxima source file: '/tmp/gen/maxima/Fancy_transforms6D' (did you forget to generate it?)
	at iit.dsl.coord.generator.maxima.converter.MaximaConverter.setModel(Unknown Source)
	at iit.dsl.coord.generator.maxima.converter.Utils.getConversionInfo(Unknown Source)
	at iit.dsl.coord.generator.cpp.LocalClasses.localClassesDefinitions(Unknown Source)
	at iit.dsl.coord.generator.cpp.ContainerClass.sourceFile(Unknown Source)
	at iit.dsl.generator.cpp.kinematics.Transforms.generate(Unknown Source)
	at iit.dsl.generator.cpp.Generator.generateTransforms(Unknown Source)
	at iit.robcogen.KinDSLWrapper.generate(Unknown Source)
	at iit.robcogen.KinDSLWrapper.generate(Unknown Source)
	at iit.robcogen.Generator.main(Unknown Source)
```

could you help me?