ethz-adrl/control-toolbox #167: What is CPPADCG in CMakelist in ct_optcon

JacksinU, 2020-09-30:
Hello. It seems that I need CPPAGCG to compile LQR file in the example file. When I comment the sentence, it remainds me that ct::core::ADCGScalar is not a member of 'ct::core'.

Is it really necessary to provide CPPAGCG if I only need lqrSolver and define the matrix A,B,Q,,R  myself?

[image]

JacksinU, 2020-10-01:
I find out that cppagcg is not necessary when I try LQR. The LQR is proposed for continuous systems but I'd like to know how to obtain LQR gain matrix in discrete state equation.

markusgft, 2020-10-31:
Hi @JacksinU 
Sorry for not addressing your issue earlier. 
Could you please share more details about your system? E.g. what are the dynamics that you are trying to stabilize? What are your cost function matrices?
Cheers!

JacksinU, 2020-11-01:
> Hi @JacksinU
> Sorry for not addressing your issue earlier.
> Could you please share more details about your system? E.g. what are the dynamics that you are trying to stabilize? What are your cost function matrices?
> Cheers!

Hello, that's fine. I realized that increasing the maximum number can solve this problem. But it takes a lot of time to obtain the solution.

I was trying to stabilize the vehicle dynamics with preivew distance attached below and aim to minimize these four states and the input in a quadratic form.

Best!
[image]
[image]

markusgft, 2020-11-01:
Hi! What exactly do you mean by "it takes a lot of time"? Seconds, Minutes?
You may also try to use the direct rather than the iterative solver, maybe that is faster in your case.
Also, did you compile with optimization?

JacksinU, 2020-11-01:
> Hi! What exactly do you mean by "it takes a lot of time"? Seconds, Minutes?
> You may also try to use the direct rather than the iterative solver, maybe that is faster in your case.
> Also, did you compile with optimization?

It is about 1 second. Since it runs in real time, 20ms or 30ms is accepted but 1s is too larger than expected. 
Sorry, what's the " direct"? And how to complie with optimization? I just complied as mentioned in quick start.
Best.

markusgft, 2020-11-01:
Ah sorry, the direct solve is only available for the CARE.

But, in order to speed up your computation times, are you already employing the warm-start capability of the iterative DARE solver? This [method](https://github.com/ethz-adrl/control-toolbox/blob/af3ea074c8d067393eeda70e0dc016967be70ddc/ct_optcon/include/ct/optcon/lqr/riccati/DARE-impl.hpp#L33) returns the converged P matrix, which you can use to warm-start the solver in your next call. Are already using this?

JacksinU, 2020-11-06:
> Ah sorry, the direct solve is only available for the CARE.
>
> But, in order to speed up your computation times, are you already employing the warm-start capability of the iterative DARE solver? This [...]
 
Sorry for the delayed reply. I have already used this function to solve the problem and employ it in every control loop. But how to warm start the solver exactly?
[image]

markusgft, 2020-11-08:
@JacksinU, there is a second version of "computeSteadyStateRiccatiMatrix", which is linked in my last post. That version takes the P from the previous solve as additional input argument and uses is as warm-start. So all you have to do is to book-keep your latest P and use it in the next iteration for the warm start. Does that work for you?

JacksinU, 2020-11-08:
> @JacksinU, there is a second version of "computeSteadyStateRiccatiMatrix", which is linked in my last post. That version takes the P from the previous solve as additional input argument and uses is [...]
 
It seems that it doesn't make a big difference even if the warm start is adopted. I store the P as a global metrix and compute it as follows.
[image]

markusgft, 2020-11-08:
No, the way you posted above is a local variable, of course that will not
work!

JacksinU, 2020-11-08:
> No, the way you posted above is a local variable, of course that will not work!
> […](#)

Sorry, I didn't post the whole figure. The matrix is indeed claried at the beginning of the file.
[image]

markusgft, 2020-11-08:
You still write in a temporary local P , not the global P. The temporary
overrides the global. Just remove the redeclaration within the curly
brackets!

JacksinU, 2020-11-08:
> You still write in a temporary local P , not the global P. The temporary overrides the global. Just remove the redeclaration within the curly brackets!
> […](#)

Oh, thanks. It works and runs faster with the warm start. It can reach 18hz now but could it achieve 30hz?

markusgft, 2020-11-08:
One possibility would be to limit the number of iterations / simply allow for a larger tolerance, e.g. 1e-5. Cheers and good luck!

closed 2020-11-10 by markusgft