ethz-adrl/control-toolbox #169: robcogen  error when  generating code

csufangyu, 2020-10-29:
hi I get the problem,but i can not slove it.I try to 0.5.0,0.5.1,and it not usefull,could you help me?thanks very much!

```
robot203@robot203:~/robcogen-0.5.0/exe$ ./robcogen.sh ../sample/model/fancy.kindsl 
 0 - CTDSL	- The coordinate transforms description file .ctdsl

 1 - ALLMx	- [All the Maxima targets]
 2 - TMx	- Maxima coordinate transforms
 3 - JMx	- Maxima geometric Jacobians

 4 - ALLC++	- [All the C++ targets]
 5 - CC++	- C++ common code
 6 - TC++	- C++ coordinate Transform
 7 - JC++	- C++ geometric Jacobians
 8 - DC++	- C++ common code for dynamics
 9 - IDC++	- C++ Inverse-Dynamics implementation
10 - FDC++	- C++ Forward-Dynamics implementation
11 - JSIMC++	- C++ Joint-Space-Inertia-Matrix calculation
12 - MKC++	- CMake file for C++ code

13 - ALLMt	- [All the Matlab targets]
14 - TMt	- Matlab coordinate transforms
15 - JMt	- Matlab geometric Jacobians
16 - DMt	- Matlab common code for dynamics
17 - IDMt	- Matlab Inverse-Dynamics implementation
18 - JSIMMt	- Matlab Joint-Space-Inertia-Matrix calculation

19 - SL	- [All the SL targets]
20 - SLR	- SL, robot code (includes the robot model .dyn)
21 - SLU	- SL, robot-user code

22 - RB	- [All the robot models]
23 - RB_ROS	- XML URDF format
24 - RB_Feath	- Matlab structure for Featherstone's code
25 - RB_SDFAST	- SD/FAST model descritpion

26 - Reload	- Reload the input files
27 - Quit	- Exit robcogen

What would you like to generate? Please enter the integer code:
1,4
Exception in thread "main" java.util.InputMismatchException
	at java.util.Scanner.throwFor(Scanner.java:864)
	at java.util.Scanner.next(Scanner.java:1485)
	at java.util.Scanner.nextInt(Scanner.java:2117)
	at java.util.Scanner.nextInt(Scanner.java:2076)
	at iit.robcogen.Generator.main(Unknown Source)
```

csufangyu, 2020-10-29:
we should enter 1,then enter 4.I cannot understand the example at https://robcogenteam.bitbucket.io/usage.html

closed 2020-10-29 by csufangyu

markusgft, 2020-10-31:
Have you tried running it strictly sequentially?
E.g,. first 0, then press enter, then 1, press enter, then 4, press enter?

csufangyu, 2020-10-31:
yes  thanks very much.   I can not understand the example on robcogen!-------- 原始邮件 --------发件人： Markus Giftthaler <notifications@github.com>日期： 2020年10月31日周六 傍晚6:28收件人： ethz-adrl/control-toolbox <control-toolbox@noreply.github.com>抄送： universe <csufangyu@163.com>, State change <state_change@noreply.github.com>主    题： Re: [ethz-adrl/control-toolbox] robcogen  error when  generating code  (#169)
Have you tried running it strictly sequentially?
E.g,. first 0, then press enter, then 1, press enter, then 4, press enter?