ethz-adrl/control-toolbox #171: Jacobian with mixed up rows?

awa152, 2020-10-29:
Hi,

I have a 7 DoF Arm and I am retrieving the jacobian from base to EE as follows:
~~~
// Create class instance
ct::rbd::MyRobot::Kinematics::Jacobians::Type_fr_world_J_fr_robot_link8 jacobian_instance;  

// Update and retrieve Jacobian with current Joint states
auto J = jacobian_instance.update(state.segment(0, 6));
~~~

As a result I get the following matrix print out in the terminal:
~~~
Jacobian in inital position
           0           -0    -0.706825            0            1            0  4.62412e-17
           0            1           -0           -1            0           -1            0
           1            0     0.707388            0 -0.000203673            0           -1
           0      0.25727            0    0.0245782            0        0.107            0
     0.30702            0     0.399027            0     0.106982            0            0
           0     -0.30702            0     0.472017            0        0.088            0
~~~
As from what I see, the Jacobian seems to be mixed up in rows. I would expect a Jacobian that results in a mapping like: 

![\dot{x} =   \begin{bmatrix} \dot{x}_{lin} \\ \dot{x}_{ang} \end{bmatrix} = J \dot{q}
](https://render.githubusercontent.com/render/math?math=%5Cdisplaystyle+%5Cdot%7Bx%7D+%3D+++%5Cbegin%7Bbmatrix%7D+%5Cdot%7Bx%7D_%7Blin%7D+%5C%5C+%5Cdot%7Bx%7D_%7Bang%7D+%5Cend%7Bbmatrix%7D+%3D+J+%5Cdot%7Bq%7D%0A)

But the resulting Jacobian seems to perform a mapping like:

![\dot{x} =   \begin{bmatrix} \dot{x}_{ang} \\ \dot{x}_{lin} \end{bmatrix} = J \dot{q}
](https://render.githubusercontent.com/render/math?math=%5Cdisplaystyle+%5Cdot%7Bx%7D+%3D+++%5Cbegin%7Bbmatrix%7D+%5Cdot%7Bx%7D_%7Bang%7D+%5C%5C+%5Cdot%7Bx%7D_%7Blin%7D+%5Cend%7Bbmatrix%7D+%3D+J+%5Cdot%7Bq%7D%0A)

I mean, it's not much work to change the upper three rows with the lower three rows, but I was just wondering if I am the only one that has that issue. Maybe I am retrieving the Jacobian the wrong way or it is something else.
I tried to get the Jacobian in MoveIt and it is exactly the same, just with the upper and lower rows mixed up. 

I am using RobCoGen version 0.4ad.0

Thanks for any hints

markusgft, 2020-10-31:
Dear @awa152 
Thanks for your interest in the CT and the detailed description of your issue!
Indeed you are completely right with your observation -- in the RBD-part of the CT, we follow the convention "orientation before position", therefore the stacked velocity vector looks like the second variant you described.
We follow the convention as used by RobCoGen, which itself follows the convention by Featherstone et al.
I hope this helps!
Cheers

closed 2020-11-01 by markusgft

awa152, 2020-11-02:
Thank you a lot for the explanation! I probably missed the part in the documentation where this was explained