ethz-adrl/control-toolbox #173: How to linearize the system in diffrent point and solve the ARE recursively

Rullec, 2020-11-03:
Hi everyone,
   Thank you for this excellent work of `control-box` and your time in this issue in advance. ;-)

# Background
- I'm working on the control of articulated characters in simulation environment, for the sake of application in computer graphic. Particularly, I am reproducing a paper from SIGGRAPH (http://grail.cs.washington.edu/projects/nqr/s2009/).
- In this paper, the author modelled the robot in Lagragian Dynamics, then define the generalized coordinate/velocity as the state vector, and the joint torque/contact force as the control vector. It proposed an revised LQR in order to make the robot running as close as possible to a given MOCAP data.

# Question Background
- I think `control-box` can help me on this LQR stuff. In this paper, the dynamics is defined:
[image]
- then the dynamics equation are linearlized in every frame of the MOCAP data, namely:
[image]

-And the **cost of LQR** is defined as :
[image]

- Follow a standard derivation in optimal control literature, We can get the following ARE:
[image]

- But it seems this ARE are defined recursively, which means that: I know the solution in the final (Nth) frame, and I can get N-1, N-2, ..., k, ..., 2, 1, 0's solution one by one. It is sooooooooooo different from the statement in our library `control box`, but it is called `ARE` as well .

# Questions

- I am eager to know how to linearize the system in different point under the help of `control-box`

- Would you be happy to give me some advices on how to deal with this problem with the `control-box` library? I am quite confused now.

Thanks for your time!

markusgft, 2020-11-03:
Dear @Rullec 
Thanks for your interest in the Control Toolbox and for posting this issue. In fact, at first glance the algorithm that you sketched above looks 100%  identical to a standard sequential linear quadratic optimal control algorithm. In that case, you are right here, we have implementations if iLQR, Gauss-Newton Multiple Shooting (and others).
In fact, if you can formulate your problem in terms of equations (4) and (6), and if you possibly have the analytic gradients of dynamics and cost functions available, you are good to go and you could try out one of the algorithms implemented here in the CT. A good starting point for you to look is the example ex_NLOC in ct_optcon. You need to implement your own dynamics function (for getting the dynamics derivatives you could use the SystemLinearizer-class, e.g. using numerical differentiation). The cost function above looks like being composed from quadratic terms only, which is also easy to realize using the CT.
I would recommend you to check out the examples! 
Also, if you were looking for more complex examples, you could take a look our repository ct_ros.
Best!

Rullec, 2020-11-04:
Hi Markus, thanks for your helpful, and constructive help! It teaches me a lot! XD

I'm very sorry about my weak background on optimal control. There might be some wrong ideas in the following response. If it makes you feel confused or strange, please point it out directly! And I will make it more clear ;)

- About the control algo: Under your guidence, just now I studyed the `iLQR` and `Gauss-Newton` algorithms carefully. But what does this sentence below mean? :O

> we have implementations if iLQR, Gauss-Newton Multiple Shooting (and others).

**What's the meaning of `if LQR, Gauss-Newton Multiple Shooting`?**

- About the Dynamic equation: I have implemented the analytic derivation in my system manually, and certainly I can formulate my system euqation in (4) and (6) easily. So **I can offer any gradient to `control-box`**;)

I have began to learn the examples in `control-box`;). Here please let me throw out another question:
**Can I integrate ONLY the optimal control (LQR) of `control-box` in my system**? I hope to offer the system gradient/linearization result to `control-box` and expecting to get the control policy returned. Do you think it's a good way? :D

Bests,
Xudong

markusgft, 2020-11-04:
Hi @Rullec,
my bad, that was a typo -- of course I meant "of" instead of "if" (also corrected above). So what I meant is that those algorithms are implemented in the CT.
To your second question: yes you can do this! Due to the CTs modularity, it should be straight-forward to pick only those algorithms and headers for your use-case, that you really need.
Best

Rullec, 2020-11-04:
Thanks for your help, I will try to implement it now!

closed 2020-11-10 by markusgft