ethz-adrl/control-toolbox #174: Updating the reference state in a mpc routine

awa152, 2020-11-18:
So far I am loading IntermediateCost and FinalCost with constant reference/desired states x_des from a .info file. The Endeffector of my manipulator moves in the desired reference.
Now I am trying to change the reference state in my mpc loop, doing the following
~~~
auto inst = ilqr_mpc.getSolver().getCostFunctionInstances();
inst[0]->getIntermediateTermById(0)->updateReferenceState(x_des);
inst[0]->getFinalTermById(0)->updateReferenceState(x_des);
~~~
where x_des is the new desired reference state. When I print out the new reference with 
~~~
std::cout << inst[0]->getIntermediateTermById(0)->getReferenceState() << std::endl;
~~~
I see, that the reference state indeed changed. But the simulated Endeffector of the controlled arm stays in the initial reference state and nothing happens. I am wondering if there's something I am missing here.

markusgft, 2020-11-21:
This can have multiple causes, and we need more info to debug.
 * what is the size of the cost function instances vector? Are you using a multi-threaded version? Then you need to iterate over all instances of the cost functions vector and change the reference everywhere
 * do you have a cost associated with that reference? Maybe your penalty matrix is zero - that would also explain the behaviour.

awa152, 2020-11-23:
Well, I tried to run the MPC in an endless loop without changing the mpc_mode in the mpc_settings from FIXED_FINAL_TIME to CONSTANT_RECEDING_HORIZON. After I changed that, the new reference got updated as expected. 

To the first point you mentioned, I used one Thread, but to support multi-threaded versions I updated the code to 
~~~
for (size_t j = 0; j < inst.size(); j++)
{
  inst[j]->getIntermediateTermById(intTermID)->updateReferenceState(x_des);
  inst[j]->getFinalTermById(finalTermID)->updateReferenceState(x_des);
}
~~~

closed 2020-11-23 by awa152