ethz-adrl/control-toolbox #175: Influencing the starting states for every integrator iteration

awa152, 2020-11-20:
I created the following simplified second order system, inheriting from the ControlledSystem Class:

<img src=
"https://render.githubusercontent.com/render/math?math=%5Cdisplaystyle+%5Cbegin%7Balign%2A%7D%0A%5Cddot%7Bq%7D+%3D+M%5E%7B-1%7D%28q%29+%5Ccdot+J%5E%7BT%7D%28q%29+%5Ccdot+control%0A%5Cend%7Balign%2A%7D%0A" 
alt="\begin{align*}
\ddot{q} = M^{-1}(q) \cdot J^{T}(q) \cdot control
\end{align*}
">

I am using this system in a MPC loop, therefor it gets integrated twice to receive velocity and position trajectories. To try something out, I would like to set the starting point of the velocity integration to zero, before the start of every integration. But only for the velocity, not for the position. 
Is there an easy way to do so, without having to change the code of the used integrator or writing a custom one? I am still thinking of a way to change the System class that I created to achieve this behavior, but so far I couldn't achieve that. 

Thank you for any advice

markusgft, 2020-11-21:
Hi @awa152 
In order to do that, you need to adopt a nonlinear state-space formulation, with a stacked vector of first and second derivative on the left hand side of the equation. See also https://en.wikipedia.org/wiki/State-space_representation
Then you can choose whatever combination of initial position and velocity you like.

awa152, 2020-11-23:
Thank you @markusgft for your fast response! My explanation might have been misleading. What I have implemented is the following:
~~~
    derivative(0) = state(7);
    derivative(1) = state(8);
    derivative(2) = state(9);
    derivative(3) = state(10);
    derivative(4) = state(11);
    derivative(5) = state(12);
    derivative(6) = state(13);
    derivative(7) = ddq(0);
    derivative(8) = ddq(1);
    derivative(9) = ddq(2);
    derivative(10) = ddq(3);
    derivative(11) = ddq(4);
    derivative(12) = ddq(5);
    derivative(13) = ddq(6);
~~~
where the states 0-6 correspond to the position and 7-13 to the velocity and ddq is the calculated acceleration as per formula above. This is the nonlinear state space representation that you mentioned and I can easily set any desired initial conditions for this system as I want.  
What I meant was to set the start state of the integration to zero in every iteration, which is probably best explained with some formulas. If we would use the explicit euler the integration would be: 
![\begin{align*}
\dot{q}_k &= \ddot{q}_k \cdot \Delta t + \dot{q}_{k-1} \\
q_k &= \dot{q}_k \cdot \Delta t + q_{k-1}
\end{align*}
](https://render.githubusercontent.com/render/math?math=%5Cdisplaystyle+%5Cbegin%7Balign%2A%7D%0A%5Cdot%7Bq%7D_k+%26%3D+%5Cddot%7Bq%7D_k+%5Ccdot+%5CDelta+t+%2B+%5Cdot%7Bq%7D_%7Bk-1%7D+%5C%5C%0Aq_k+%26%3D+%5Cdot%7Bq%7D_k+%5Ccdot+%5CDelta+t+%2B+q_%7Bk-1%7D%0A%5Cend%7Balign%2A%7D%0A)
What I found in a paper and what I am trying to implement is the following:
![\begin{align*}
\dot{q}_k &= \ddot{q}_k \cdot \Delta t \\
q_k &= \dot{q}_k \cdot \Delta t + q_{k-1}
\end{align*}
](https://render.githubusercontent.com/render/math?math=%5Cdisplaystyle+%5Cbegin%7Balign%2A%7D%0A%5Cdot%7Bq%7D_k+%26%3D+%5Cddot%7Bq%7D_k+%5Ccdot+%5CDelta+t+%5C%5C%0Aq_k+%26%3D+%5Cdot%7Bq%7D_k+%5Ccdot+%5CDelta+t+%2B+q_%7Bk-1%7D%0A%5Cend%7Balign%2A%7D%0A)
where they basically pretend that the systems starts at zero velocity in every integration iteration. 
I now managed to implement this by just simply setting the derivative of the position to:
![\begin{align*}
q_k &= \ddot{q}_k \cdot \Delta t^2 + q_{k-1}
\end{align*}
](https://render.githubusercontent.com/render/math?math=%5Cdisplaystyle+%5Cbegin%7Balign%2A%7D%0Aq_k+%26%3D+%5Cddot%7Bq%7D_k+%5Ccdot+%5CDelta+t%5E2+%2B+q_%7Bk-1%7D%0A%5Cend%7Balign%2A%7D%0A)
and retrieving the velocity state by finite differences.
As to why exactly they are doing it like this I am clueless. They are not really interested in simulating the real dynamics of the manipulator, but more in receiving a smooth and stable result for the position trajectory and apparently doing it like this makes the system more stable and doesn't result in overshooting, because, from what I see, you basically turn a second order system into a first order system. 
As I found a solution for the implementation I will close the issue, but if you have any thoughts on what kind of effects it has to set the velocity to zero in every integration, I would be glad to hear them.

closed 2020-11-23 by awa152