ethz-adrl/control-toolbox #176: Build Error for ct_models

agutte, 2020-11-27:
Hi, 

I'm using Ubuntu 18.04 LTS. When I try to build the control-toolbox with catkin i get the following Error messages: 

`Errors     << ct_models:make /home/alex/workspace/logs/ct_models/build.make.002.log                                                                                    
/home/alex/workspace/src/external/control-toolbox/ct_models/src/Irb4600/transform6d.cpp: In function ‘void irb4600_ik::ComputeFk(const IkReal*, irb4600_ik::IkReal*, irb4600_ik::IkReal*)’:
/home/alex/workspace/src/external/control-toolbox/ct_models/src/Irb4600/transform6d.cpp:301:242: warning: variable ‘x61’ set but not used [-Wunused-but-set-variable]
 IkReal x0,x1,x2,x3,x4,x5,x6,x7,x8,x9,x10,x11,x12,x13,x14,x15,x16,x17,x18,x19,x20,x21,x22,x23,x24,x25,x26,x27,x28,x29,x30,x31,x32,x33,x34,x35,x36,x37,x38,x39,x40,x41,x42,x43,x44,x45,x46,x47,x48,x49,x50,x51,x52,x53,x54,x55,x56,x57,x58,x59,x60,x61,x62,x63,x64;
                                                                                                                                                                                                                                                  ^~~
In file included from /home/alex/workspace/devel/include/ct/core/core.h:23:0,
                 from /home/alex/workspace/src/external/control-toolbox/ct_models/include/ct/models/InvertedPendulum/codegen/InvertedPendulumActDynLinearizedForward.h:8,
                 from /home/alex/workspace/src/external/control-toolbox/ct_models/include/ct/models/InvertedPendulum/codegen/InvertedPendulumActDynLinearizedForward.cpp:6:
/home/alex/workspace/devel/include/ct/core/internal/autodiff/CppadParallel.h:64:10: error: ‘map’ in namespace ‘std’ does not name a template type
     std::map<size_t, size_t> threadMap_;
          ^~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [CMakeFiles/InvertedPendulumActDynLinearizedForward.dir/include/ct/models/InvertedPendulum/codegen/InvertedPendulumActDynLinearizedForward.cpp.o] Error 1
make[1]: *** [CMakeFiles/InvertedPendulumActDynLinearizedForward.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /home/alex/workspace/devel/include/ct/core/core.h:23:0,
                 from /home/alex/workspace/src/external/control-toolbox/ct_models/include/ct/models/HyA/codegen/HyALinearizedForward.h:8,
                 from /home/alex/workspace/src/external/control-toolbox/ct_models/include/ct/models/HyA/codegen/HyALinearizedForward.cpp:6:
/home/alex/workspace/devel/include/ct/core/internal/autodiff/CppadParallel.h:64:10: error: ‘map’ in namespace ‘std’ does not name a template type
     std::map<size_t, size_t> threadMap_;
          ^~~
compilation terminated due to -Wfatal-errors.
make[2]: *** [CMakeFiles/HyALinearizedForward.dir/include/ct/models/HyA/codegen/HyALinearizedForward.cpp.o] Error 1
make[1]: *** [CMakeFiles/HyALinearizedForward.dir/all] Error 2
make: *** [all] Error 2
`

To the best of my knowledge I installed all dependencies needed. Do you have an idea what the problem might be?

markusgft, 2020-11-27:
Hi @agutte,
thanks for reporting your issue. I have not seen it before.
Could you simply try adding the include

```
#include <map>
```
to the file `ct/core/internal/autodiff/CppadParallel.h`?

A long shot...another guess would be that you have a second (oder, probably incompatible) version of CppAD installed on your system, which gets found by cmake and causes that error... could you verify that this is not the case?

agutte, 2020-11-30:
Hi @markusgft,
thanks for your quick reply! 

Including <map> was not solving the problem, as similar error messages raised as a consequence . However, it led me to another error with the CppAD package, as you were suspecting. 

After a bit of trouble installing the newest version, I just reinstalled CppAD and the build of the control-toolbox succeeded. 
Thanks for your help!

closed 2020-11-30 by agutte