ethz-adrl/control-toolbox #177: Constraining the acceleration in a second order system

awa152, 2020-11-30:
When defining a second order system in the CT, where I compute the acceleration based on position and velocity, I define position and velocity as states. When solving an optimal control problem for such a system, the states can be easily constrained and also retrieved for plotting. I was wondering if the same goes for the acceleration. 
I know I could get the acceleration with the resulting trajectories for position and velocity after the optimization is finished, but I would like to know if I am able to also constrain the acceleration while finding the solution. Do I have to define the acceleration as a state as well? And if so, how exactly am I supposed to do so?

awa152, 2021-01-13:
I am still curious if there is a way to achieve this. I have implemented the system as follows:
~~~
    derivative(0) = state(7);
    derivative(1) = state(8);
    derivative(2) = state(9);
    derivative(3) = state(10);
    derivative(4) = state(11);
    derivative(5) = state(12);
    derivative(6) = state(13);
    derivative(7) = ddq(0);
    derivative(8) = ddq(1);
    derivative(9) = ddq(2);
    derivative(10) = ddq(3);
    derivative(11) = ddq(4);
    derivative(12) = ddq(5);
    derivative(13) = ddq(6);
~~~
where ddq are my joint accelerations that get computed with forward dynamics and then being integrated twice by the methods provided with the control toolbox. Setting upper and lower bounds for the states 0-13 in the statevector is easily achieved following the example: 

https://ethz-adrl.github.io/ct/ct_doc/doc/html/optcon_tut_nloc_box_constrained.html

But what about ddq? Is there a way to specify upper and lower bounds for ddq using the control toolbox? I was thinking about defining ddq as new states using finite differences, by just saving the old value when the computeControlledDynamics method of my system class gets called. But this does not work, as I don't know how often the mehtod gets called, as this depends on different factors as the integration method that is used and probably other things as well. 

@markusgft I am just curious if this is generally possible using the control toolbox and also if there is an easy way to achieve this, that I might missed.

markusgft, 2021-01-16:
Hi @awa152, 

constrainting the acceleration is very straightforward if you define your system as a simple double-integrator, as you do in your above code example. Using the constrained LQ-solver HPIPM, you can define a box-constraint for the control input. Since the control input is the acceleration, you can box-constrain acceleration. 
You should be able to directly derive that from https://ethz-adrl.github.io/ct/ct_doc/doc/html/optcon_tut_nloc_box_constrained.html

If your control input is a robot joint torque, things are more difficult - you would have to introduce a general nonlinear inequality constraint which includes the system forward dynamics. If you are interested in a simple solution, I would not recommend that for now -- instead, I would recommend to use joint acceleration as inputs, and then reconstruct the robot joint torque via inverse dynamics.

Best!
Markus

awa152, 2021-01-18:
Thank you for your answer @markusgft,

unfortunately my control input isn't ddq, but a six dimensional force vector. The simplified forward dynamics I am using for my system to compute ddq is: ddq = M⁻1 * J^T * control. I would like to keep the force vector as input, so I am probably bound to use a general nonlinear inequality constraint. Is there an example file where such a constraint is implemented, that I could look into?

Kind regards!

markusgft, 2021-01-25:
Yes, there are corresponding examples in ct/optcon, ex_NLOC_general_constrained.cpp
But please be aware that this is a disadvantageous formulation what you are aiming for, since all your system nonlinearity will be reflected in the general constraint. 
If you are interested in making your solution straight-forward and more robust, I would recommend to opt for acceleration as control input, use the box constraint for bounding acceleration, and then employ inverse dynamics to compute the resulting joint torque. If you use a quadratic cost function in this scenario, you will end up with a classical box-constrained LQ optimal control problem, which the CT-solvers are guaranteed to solve. If you go down the complex route, you get a general nonlinear, nonlinearly constrained optimal control problem, which may not be solvable at all using iterative optimal control.

awa152, 2021-01-27:
Thank you very much for your answer @markusgft,
I tried using the acceleration as input and it works just fine! You were right, going down the complex route and using general nonlinear constraints is probably not the smartest idea if there is an easier solution to this. 

I think I can close the issue now. Thanks for helping!

closed 2021-01-27 by awa152