ethz-adrl/control-toolbox #179: robcogen JacomaxConfigurationException

EpicDuckPotato, 2021-01-02:
Hi,

I used urdf2robcogen to create kindsl and dtdsl files from a urdf, and now I'm trying to generate c++ code from these files. However, I get the following error:

Exception in thread "main" uk.ac.ed.ph.jacomax.JacomaxConfigurationException: Could not obtain a MaximaConfiguration after attempting configuration methods [PROPERTIES_SEARCH, AUTO]
at uk.ac.ed.ph.jacomax.JacomaxSimpleConfigurator.configure(JacomaxSimpleConfigurator.java:92)
at uk.ac.ed.ph.jacomax.JacomaxSimpleConfigurator.configure(JacomaxSimpleConfigurator.java:46)
at iit.dsl.maxdsl.utils.MaximaRunner.<init>(Unknown Source)
at iit.dsl.coord.generator.maxima.converter.MaximaConverter.setModel(Unknown Source)
at iit.dsl.coord.generator.maxima.converter.Utils.getConversionInfo(Unknown Source)
at iit.dsl.coord.generator.cpp.LocalClasses.localClassesDefinitions(Unknown Source)
at iit.dsl.coord.generator.cpp.ContainerClass.sourceFile(Unknown Source)
at iit.dsl.generator.cpp.kinematics.Transforms.generate(Unknown Source)
at iit.dsl.generator.cpp.Generator.generateTransforms(Unknown Source)
at iit.robcogen.KinDSLWrapper.generate(Unknown Source)
at iit.robcogen.KinDSLWrapper.generate(Unknown Source)
at iit.robcogen.Generator.main(Unknown Source)

I'm simply following the instructions on the website and running

./robcogen.sh /path/to/kindsl /path/to/dtdsl

Any idea what's going on?

Thanks,
Anoop Bhat

markusgft, 2021-01-03:
Dear Anoop,
it looks like there is a problem with maxima. Do you have maxima installed on your system?
$ sudo apt-get install maxima
Also, are you sure you are exporting all maxima targets in the first place? typically you need to create all maxima targets before creating the C++ targets.
Are you using the AD-compatible version of robcogen?

EpicDuckPotato, 2021-01-03:
Thanks for the response! Turns out that I just needed to install maxima, and I could generate code without errors with robcogen-0.5.2. Will I be able to use 0.5.2's automatic differentiation capabilities with control-toolbox?

The documentation (https://ethz-adrl.github.io/ct/ct_doc/doc/html/rbd_tut_modelling.html) recommends robcogen-0.4ad.0, though, and if I use that, I get the following error:

Exception in thread "main" java.lang.NullPointerException
	at iit.dsl.maxdsl.utils.MaximaConversionUtils.cleanMatrixRowOutput(Unknown Source)
	at iit.dsl.maxdsl.utils.MaximaConversionUtils.getMatrix(Unknown Source)
	at iit.dsl.coord.generator.maxima.converter.MaximaConverter.parseTransform(Unknown Source)
	at iit.dsl.coord.generator.cpp.TransformsClasses.localClassesDefinitions(Unknown Source)
	at iit.dsl.coord.generator.cpp.EigenFiles.implementationsFileContent(Unknown Source)
	at iit.dsl.generator.cpp.kinematics.Transforms.generate(Unknown Source)
	at iit.dsl.generator.cpp.Generator.generateTransforms(Unknown Source)
	at iit.robcogen.KinDSLWrapper.generate(Unknown Source)
	at iit.robcogen.KinDSLWrapper.generate(Unknown Source)
	at iit.robcogen.Generator.main(Unknown Source)

awa152, 2021-01-12:
The NullpointerException when using 0.4ad.0 was addressed in issue https://github.com/ethz-adrl/control-toolbox/issues/166

Simply removing a "/" sign from the framework.properties configuration resolves the issue, like:
generator.maxima.libs.path = ../etc/maxima-libs (do NOT use "/" at the end).

EpicDuckPotato, 2021-01-12:
> The NullpointerException when using 0.4ad.0 was addressed in issue #166
> 
> Simply removing a "/" sign from the framework.properties configuration resolves the issue, like:
> generator.maxima.libs.path = ../etc/maxima-libs (do NOT use "/" at the end).

This worked, thanks!

closed 2021-01-12 by EpicDuckPotato