ethz-adrl/control-toolbox #182: Terminal Constraints and Soft Constraints

awa152, 2021-01-27:
Hi, I have just a few general question about the control toolbox.

1. In MPC theory the use of terminal constraints is quite common, where the state at the end of the prediction horizon should be constraint to an output admissible set Xf. I see that the ct has a TerminalConstraint.h term, but it seems this is limited to an equality constraint for the state. Does the ct provide the possibility to create a terminal constraint, where the state at the end of the prediction horizon is an element of a set Xf?
2. Is there a way to use soft constraints, i.e. through introduction of slack variables? 

Thanks in advance

markusgft, 2021-01-27:
Hi, yes, we support terminal box and general terminal contraints. Their use is analoguous to the intermediate constraints in the examples that we discussed earlier.
We currently only support hard constraints, soft constraints are theoretically exposed in hpipm, but not interfaced to the CT!
Best

awa152, 2021-02-02:
Thank you @markusgft 

and what about extra terms for the cost function? 

So far I create two `ct::optcon::TermQuadratic` terms, which define intermediate and final cost for my MPC. Then I create a `ct::optcon::CostFunctionAnalytical` object, where I add those terms. Basically the same cost function configuration as in the example: https://ethz-adrl.github.io/ct/ct_doc/doc/html/optcon_tut_nloc.html. Also I update my costfunction instances with the desired state vector.

So I should end up with a costfunction that looks like:
<img src=
"https://render.githubusercontent.com/render/math?math=%5Cdisplaystyle+%5Cbegin%7Balign%2A%7D%0AI+%3D+%5Csum%5E%7BN-1%7D_%7Bk%3D0%7D%7B%28x_%7Bdes%7D+-+x_k%29%5ETQ_%7Bstage%7D%28x_%7Bdes%7D+-+x_k%29+%2B+u_k%5ETR_%7Bstage%7Du_k%7D+%2B+%28x_%7Bdes%7D+-+x_N%29%5ETQ_%7Bfinal%7D%28x_%7Bdes%7D+-+x_N%29+%2B+u_N%5ETR_%7Bfinal%7Du_N%0A%5Cend%7Balign%2A%7D%0A" 
alt="\begin{align*}
I = \sum^{N-1}_{k=0}{(x_{des} - x_k)^TQ_{stage}(x_{des} - x_k) + u_k^TR_{stage}u_k} + (x_{des} - x_N)^TQ_{final}(x_{des} - x_N) + u_N^TR_{final}u_N
\end{align*}
">

In my case my states are joint position, joint velocity as well as the cartesian pose, defined through:
<img src=
"https://render.githubusercontent.com/render/math?math=%5Cdisplaystyle+%5Cbegin%7Balign%2A%7D%0A%5Cdot%7Bx%7D+%3D+J%5Cdot%7Bq%7D%0A%5Cend%7Balign%2A%7D%0A" 
alt="\begin{align*}
\dot{x} = J\dot{q}
\end{align*}
">

The aim is, to move the endeffector to the desired cartesian pose. After each iteration I update the current pose with the forward dynamics, which works quite well with the position, but not the orientation. As the differential equation for the pose gives me the angular velocity, I somehow try to retrieve euler angles from the rotation matrix I get from the `getEERotInBase` method. Due to the nature of euler angles, there are quite some issues with this approach. 

There are other ways of defining an orientation error using rotation matrices or unit quaternions, that I would like to try. My problem is, that I am not quite sure how to realize this inside the control toolbox. For example I would like to compute an error:
<img src=
"https://render.githubusercontent.com/render/math?math=%5Cdisplaystyle+%5Cbegin%7Balign%2A%7D%0Ae_O%3D+0.5%28n_c%28q%29+%5Ctimes+n_d+%2B+s_c%28q%29+%5Ctimes+s_d+%2B+a_c%28q%29+%5Ctimes+a_d%29%0A%5Cend%7Balign%2A%7D%0A" 
alt="\begin{align*}
e_O= 0.5(n_c(q) \times n_d + s_c(q) \times s_d + a_c(q) \times a_d)
\end{align*}
">
based on the current and desired columns of the rotation matrices. I would like to be able to weight this error inside my cost function with a term:
<img src=
"https://render.githubusercontent.com/render/math?math=%5Cdisplaystyle+%5Cbegin%7Balign%2A%7D%0Ae_O%5ETQ_%7BO%7De_O%0A%5Cend%7Balign%2A%7D%0A" 
alt="\begin{align*}
e_O^TQ_{O}e_O
\end{align*}
">

And sorry for bothering you so much, my c++ skills are quite limited, so I have troubles finding my way through the ct source code.

awa152, 2021-02-03:
Hi,

I think I slowly get the gist of it. To implement a custom costfunction term as I stated above, I have to derive from the `TermBase class`. What I am trying to do with the orientation error, you basically already defined in the costfunction term in the `rgb module`: 
https://github.com/ethz-adrl/control-toolbox/blob/v3.0.2/ct_rbd/include/ct/rbd/robot/costfunction/TermTaskspaceGeometricJacobian.hpp
where you define an error based on an error quaternion and AngleAxis. Now I am trying to define my custom Term based on this cost function term. As I am not using one of the predefined systems, but my custom system, I probably have to adjust a few things, so the methods `evaluate`, `stateDerivative` and `stateSecondDerivative` know where the joint positions inside my state vector x are. But there was already a problem that I encountered on the way. 

Initializing and adding my custom costfunction term to my custfunction worked so far 
~~~
    std::shared_ptr<CustomTermTaskspaceGeometricJacobian<ct::rbd::MyRobot::Kinematics, STATE_DIM, CONTROL_DIM>>
        termTaskSpace(new CustomTermTaskspaceGeometricJacobian<ct::rbd::MyRobot::Kinematics, STATE_DIM, CONTROL_DIM>());

    termTaskSpace->loadConfigFile(cost_file, "termTaskSpace", verbose);

    m_intermediateTermTaskSpaceID_ = costFunction->addIntermediateTerm(termTaskSpace);
~~~

The problem comes when trying to update the costfunction term. The costfunction term as you defined it, has the methods `setReferencePosition` and `setReferenceOrientation` with which I would like to update the desired Pose. But using:
~~~
  // Get Cost Function instances
  auto inst = iLQR.getCostFunctionInstances();

  // Update Reference in Costfunction instances with new target pose
  for (size_t j = 0; j < inst.size(); j++)
  {
      inst[j]->getIntermediateTermById(m_intermediateTermTaskSpaceID_)->setReferencePosition(x_pos);
  }
~~~
results in the error:
~~~
error: ‘using element_type = class ct::optcon::TermBase<20, 7, double, double> {aka class ct::optcon::TermBase<20, 7, double, double>}’ has no member named ‘setReferencePosition’; did you mean ‘getReferenceState’?
       inst[j]->getIntermediateTermById(m_intermediateTermTaskSpaceID_)->setReferencePosition();
                                                                         ^~~~~~~~~~~~~~~~~~~~
                                                                         getReferenceState
~~~

The `getIntermediateTermById` method inside the `CostFunctionQuadratic` class returns a pointer to `TermBase`. As TermBase has no idea that `setReferencePosition` method of the derived class exists, the error occurs. Now I am confused on how to update this term. Overriding the `updateReferenceState` method of `TermBase` class seems like an option, but I am not quite sure.

awa152, 2021-02-03:
The solution to this was using a `std::dynamic_pointer_cast` so the pointer to `TermBase` is set to a pointer to my custom term `CustomTermTaskspaceGeometricJacobian`. The update now looks like this 
~~~
  // Get Cost Function instances
  auto inst = m_mpc_->getSolver().getCostFunctionInstances();

  // Update Reference in Costfunction instances with new target pose
  for (size_t j = 0; j < inst.size(); j++)
  {
      // Pointer to TermBase
      auto taskSpaceTerm = inst[j]->getIntermediateTermById(m_intermediateTermTaskSpaceID_);

      // Pointer cast to custom Term which derives from TermBase
      std::shared_ptr<termTaskSpace_t> derived = std::dynamic_pointer_cast<termTaskSpace_t>(taskSpaceTerm);

      // Update Costfunction Term
      derived->setReferencePosition(m_targetpose_.getTargetPosition());
      derived->setReferenceOrientation(m_targetpose_.getTargetQuaternion());
  }
~~~

In my `CustomTermTaskspaceGeometricJacobian` class I also had to make following changes:
~~~
rbdState.jointPositions() = x.segment(0, KINEMATICS::NJOINTS - 1);  // x.template head<KINEMATICS::NJOINTS>();
~~~
inside the `evaluate`, `stateDerivative` and `stateSecondDerivative` methods, as I am using my custom system and I have to tell the methods where inside my state vector the joint positions are located.

closed 2021-02-03 by awa152