ethz-adrl/control-toolbox #183: building other examples

BarzinM, 2021-02-03:
Hi, sorry if this question is too basic. How can I build other ct_optcon examples such as `control-toolbox/ct_optcon/examples/NLOC_MPC.cpp`?
When I build the project, it automatically builds `ex_NLOC_boxConstrained` and `ex_NLOC_generalConstrained` but other examples files such as NLOC_MPC don't get built.
I'm *not* using catkin system.

markusgft, 2021-02-08:
Hi, probably you need to build with `-DBUILD_EXAMPLES=true` !

justinberi, 2021-02-12:
Hi,
I was following the [QuickStart](https://github.com/ethz-adrl/control-toolbox/wiki/Quickstart) instructions using catkin tools.

For `ex_NLOC_boxConstrained` not building.
HPIPM was not installed automagically. I had to run `ct/install_hpimp.sh`.
Running that script fixed the problem.

For `ex_NLOC` not building.
I noticed this executable (and more) are commented out in `ct_optcon/examples/CMakeLists.txt`.
Adding them back fixed that problem.

BarzinM, 2021-02-24:
yes, thanks

closed 2021-02-24 by BarzinM