ethz-adrl/control-toolbox #184: Control Horizon

awa152, 2021-02-08:
I am just interested if there is a possibility to specify a control horizon, where the last control value calculated is hold constant for the rest of the prediction horizon. To my knowledge so far the implementation makes the assumption that: 
control horizon = prediction horizon

Best

markusgft, 2021-02-08:
Yes you are correct, that is the assumption. If you really wanted to implement your above mentioned feature, you could modify the code such that the last control command gets "held" for a longer time. I doubt, however, that the benefit will be large.

awa152, 2021-02-09:
Thank you for the fast answer! I was just interested if this possibility exists, as it seems that this is also a quite common thing to do.

closed 2021-02-09 by awa152