ethz-adrl/control-toolbox #186: Stopping solver when total cost is below certain threshold

awa152, 2021-02-25:
I am encountering some interesting behavior, when solving my problem in an receding horizon MPC routine. 
When I update a new target pose, the first solution with GNMS takes 4 iterations and is shown below. 
For that I set printSummary and debugPrint to true. 

~~~
[MultipleShooting]: computing LQ Approximation from index 5 to N-1 took 1.547365 ms
[MultipleShooting]: Solving prepare stage of LQOC Problem
[MultipleShooting]: Prepare phase of LQOC problem took 0.002475 ms
[MultipleShooting]: prepareIteration() took 1.802402 ms
[MultipleShooting]: computing LQ approximation for first multiple-shooting interval took 0.184508 ms
[MultipleShooting]: Finish phase LQOC Problem
[MultipleShooting]: Finish solving LQOC problem took 7.557204 ms
[MultipleShooting]: Line search took 0.310682 ms
[MultipleShooting]: finishIteration() took 8.339397 ms
NLOC Summary of iteration 0
==============================
interm. cost:	1.37565964361
final cost:	0
total cost:	1.37565964361
total merit:	1.37565964361
tot. defect L1:	0.212041106208
tot. defect L2:	0.0973813909403
box cstr.err:	0
gen cstr.err:	0
total lx norm:	26.319324674
total lu norm:	14.9739679897
step-size:	1
                   ===========

[MultipleShooting]: computing LQ Approximation from index 5 to N-1 took 0.608909 ms
[MultipleShooting]: Solving prepare stage of LQOC Problem
[MultipleShooting]: Prepare phase of LQOC problem took 0.00051 ms
[MultipleShooting]: prepareIteration() took 0.635876 ms
[MultipleShooting]: computing LQ approximation for first multiple-shooting interval took 0.056361 ms
[MultipleShooting]: Finish phase LQOC Problem
[MultipleShooting]: Finish solving LQOC problem took 5.166026 ms
[MultipleShooting]: Line search took 0.329641 ms
[MultipleShooting]: finishIteration() took 5.602459 ms
NLOC Summary of iteration 1
==============================
interm. cost:	1.31085060474
final cost:	0
total cost:	1.31085060474
total merit:	1.31085060474
tot. defect L1:	0.0141449671345
tot. defect L2:	0.00663556822773
box cstr.err:	0
gen cstr.err:	0
total lx norm:	5.41752561809
total lu norm:	2.80196187147
step-size:	1
                   ===========

[MultipleShooting]: computing LQ Approximation from index 5 to N-1 took 1.163349 ms
[MultipleShooting]: Solving prepare stage of LQOC Problem
[MultipleShooting]: Prepare phase of LQOC problem took 0.0083 ms
[MultipleShooting]: prepareIteration() took 1.25171 ms
[MultipleShooting]: computing LQ approximation for first multiple-shooting interval took 0.132047 ms
[MultipleShooting]: Finish phase LQOC Problem
[MultipleShooting]: Finish solving LQOC problem took 9.610376 ms
[MultipleShooting]: Line search took 0.325102 ms
[MultipleShooting]: finishIteration() took 10.132516 ms
NLOC Summary of iteration 2
==============================
interm. cost:	1.30993248119
final cost:	0
total cost:	1.30993248119
total merit:	1.30993248119
tot. defect L1:	0.000102134243653
tot. defect L2:	5.92240828319e-05
[... 346 lines cut ...]
    epsilon 0
    max_iterations 3
    fixedHessianCorrection false
    recordSmallestEigenvalue false
    min_cost_improvement 1e-5
    maxDefectSum 1e-5
    meritFunctionRho 0.0
    meritFunctionRhoConstraints 0.0
    nThreads 1
    nThreadsEigen 1
    printSummary true 
    debugPrint true
    ; logToMatlab 0   
    
    line_search
    {
        type NONE
        adaptive false
        maxIterations 10
        alpha_0 1.0
        n_alpha 0.5
        debugPrint true
    }
}
~~~

markusgft, 2021-02-25:
Hi, my recommendation is to run only a single iteration in every mpc cycle!

awa152, 2021-02-26:
Hi @markusgft,
thank you for the fast answer. If we don't consider the computational effort for a moment, wouldn't be a single iteration to little for the solver to come close to an optimal solution? 

Also I noticed that despite giving the MPC class a NLOptConSettings instance with a max_iterations count of greater than one, it always seems to make only a single iteration. At least it always prints out an Iteration count of 0 when I set printSummary to true. I don't know if this is intended like this. 
That's why I use NLOptConSolver directly and if needed the solver takes the amount of iterations set in max_iterations.

markusgft, 2021-02-26:
Doing only iteration is a very Standard approach in MPC. You can use the additional bandwidth to send controller updates more frequently

awa152, 2021-02-26:
Thank you for the input @markusgft

closed 2021-02-26 by awa152