ethz-adrl/control-toolbox #188: ct_core build is failing

vvrs, 2021-03-28:
I have `control-toolbox` and `kindr` modules in `src`, when did `catkin build -DCMAKE_BUILD_TYPE=Release` the build is failing with the following error. 

```
/usr/local/include/cppad/cg/model/dynamic_lib/dynamic_library_processor.hpp:84:16: error: cannot convert ‘const std::unique_ptr<const std::__cxx11::basic_string<char> >’ to ‘const string*’ {aka ‘const std::__cxx11::basic_string<char>*’} in return
   84 |         return _customLibExtension;
      |                ^~~~~~~~~~~~~~~~~~~
      |                |
      |                const std::unique_ptr<const std::__cxx11::basic_string<char> >
```

Any help resolving this issue is much appreciated.

markusgft, 2021-04-05:
Very likely you have the wrong versions of cppad and cppad-cg installed. Please check out the versions provided in the install script!

vvrs, 2021-04-07:
Thanks for the respose. That worked.

closed 2021-04-07 by vvrs