ethz-adrl/control-toolbox #193: RobCoGen unexpected number of elements in matrix + J

Nickick-ICRS, 2021-05-10:
Hi! I've hit a problem trying to generate the RobCoGen files. At first I got everything to generate fine, but I was using 0.5.2, which caused issues as the tpl namespace didn't exist.

Switching to the expected 0.4ad.0 and readjusting config parameters (including the fix in issue #166 ), I can generate everything except the Jacobians. This issue persists whether I try to generate full Jacobians from e.g. my world frame to my end effector, or whether I split them up (the way I have the transforms split up - world_x_a and a_x_end_effector etc.)

The exception I'm getting after running 0, 1, 4 (or 0, 1, 5, 6, 7) is:
```
Exception in thread "main" java.lang.RuntimeException: Unexpected number of elements in row 5 of the matrix + J. Expected 5, found 7
	at iit.dsl.maxdsl.utils.MaximaConversionUtils.getMatrix(Unknown Source)
	at iit.dsl.generator.maxima.Converter.getJacobianText(Unknown Source)
	at iit.dsl.generator.cpp.kinematics.Jacobians.localClassesDefinitions(Unknown Source)
	at iit.dsl.generator.cpp.kinematics.Jacobians.sourceFile(Unknown Source)
	at iit.dsl.generator.cpp.Generator.generateJacobiansFiles(Unknown Source)
	at iit.robcogen.KinDSLWrapper.generate(Unknown Source)
	at iit.robcogen.KinDSLWrapper.generate(Unknown Source)
	at iit.robcogen.Generator.main(Unknown Source)
```

Any ideas what is causing this and how to fix it? My robot is kind of a 2 branched tree if that helps.

csufangyu, 2021-05-19:
I get the same  problem,have you solved it?

JuliusSustarevas, 2021-07-27:
I have a similar error, but its in matrix + trigsimp.

markusgft, 2021-07-29:
How many joints does the robot have?

JuliusSustarevas, 2021-07-29:
More than 6 :) was using wrong description, my bad. Ended up generating
fine.