ethz-adrl/control-toolbox #196: RobCoGen can generate all cpp files except for jacobians for kinova 6 axis robot

mustang66ytz, 2021-07-04:
When I was following the instructions to generate cpp code for Kinova 6-axis robot, it always fails when generating the jacobians.cpp.

I wander is this caused by the long processing time? If so, where should I change the Maxima timeout parameter and give it a try?

I would be very thankful to any suggestions because I have been stuck here for quite a while haha!

> Exception in thread "main" java.lang.RuntimeException: Maxima timeout exception
> 	at iit.dsl.maxdsl.utils.MaximaRunner.run(Unknown Source)
> 	at iit.dsl.maxdsl.utils.MaximaConversionUtils.getMatrix(Unknown Source)
> 	at iit.dsl.coord.generator.maxima.converter.MaximaConverter.toText(Unknown Source)
> 	at iit.dsl.coord.generator.maxima.converter.MaximaConverter.process(Unknown Source)
> 	at iit.dsl.generator.maxima.Converter.process(Unknown Source)
> 	at iit.dsl.generator.cpp.kinematics.Jacobians.localClassesDefinitions(Unknown Source)
> 	at iit.dsl.generator.cpp.kinematics.Jacobians.sourceFile(Unknown Source)
> 	at iit.dsl.generator.cpp.Generator.generateJacobiansFiles(Unknown Source)
> 	at iit.robcogen.KinDSLWrapper.generate(Unknown Source)
> 	at iit.robcogen.KinDSLWrapper.generate(Unknown Source)
> 	at iit.robcogen.Generator.main(Unknown Source)

Best regards

markusgft, 2021-07-05:
Hi, in the version of robcogen that we are using in the CT, robcogen should generate the -impl headers rather than *.cpp files. Are you sure you are using the linked version of robcogen, please dont use the current main version. In case of further problems, I would recommend reaching out to the developer of robcogen, Marco Frigerio.

mustang66ytz, 2021-07-06:
Hi, thanks for your advice!!
I followed the RobCoGen official tutorial and used the RobCoGen-0.5.2 version, could I know what is the "linked version" that you are referring to?
Best regards

markusgft, 2021-07-06:
https://bitbucket.org/robcogenteam/robcogen-downloads/downloads/robcogen-0.4ad.0.zip

mustang66ytz, 2021-07-27:
@markusgft It turns out that since my urdf model consists of multiple links, whose centers of mass have translation and orientation offset relative to their parent links. Therefore, the kindsl format model converted from the urdf format model generate a few more intermediate link frames on top of the original number of links, which results in a kinematic chain too long to be properly handled by Maxima. 
I think what I need to do next is to figure out how to describe the model properly to minimize the extra DOFs got involved.