ethz-adrl/control-toolbox #198: Is this Control-toolbox package take lagged states and control inputs into consideration when calculating the derivative/Jacobian?

Daniel-Ming, 2021-07-25:
Is this ct package take lagged state and control input into consideration when calculating the derivative/Jacobian? Because our system ODE has 15 lag sizes for state and control input, I'm unsure whether this package can be used to simulate our system model.

markusgft, 2021-07-25:
Hi, thanks for the interest in the CT!
Could you please detail what you exactly mean with „lagged state“? Best

Daniel-Ming, 2021-07-26:
Our system model is a NARX type model which updates the system states depends on all previous 15 steps ago states and control inputs. According to the tutorial documentation in the Derivative/Jacobian session, the control-toolbox seems capable of handling the general ODE, which updates the system state depending on the previous 1 step ago state and control input.

markusgft, 2021-07-26:
True, if you cannot rewrite your system as ODE, we cannot handle it!