ethz-adrl/control-toolbox #201: Failed to generate .kindsl model from the .urdf models provided

mustang66ytz, 2021-07-29:
Hi,
I just wondered are the .urdf models provided in the control_toolbox/ct_models/urdf/ directory can be converted to .kindsl models successfully by using the urdf2robcogen ros package? I have tried both the irb4600 model and quadrotor with load model, but neither works successfully.

The error code is like the following:
```

terminate called after throwing an instance of 'std::runtime_error'
  what():  [std::runtime_error] /home/robot-control/wbc_ws/src/kindr/include/kindr/quaternions/Quaternion.hpp:320: UnitQuaternion() Input quaternion has not unit length.
Comparing Scalars:
Scalar 1: norm() = 0
Scalar 2: static_cast<Scalar>(1.0) = 1
Error was 100% > 0.01%

```

Thanks for any suggestions

markusgft, 2021-07-29:
Hi,
I have not used urdf2robcogen in a really long time, it may well be that it is not supporting those urdfs (anymore).
I would consider it unmaintained, in doubt you could try contacting the developers!
Best

mustang66ytz, 2021-07-29:
Cool, thanks very much!