ethz-adrl/control-toolbox #202: Status of the overall project

Ravenwater, 2021-08-01:
Given the note regarding this repo being sparsely maintained, two questions:

- has this repo reached its EOL?
- has development moved to another library that is used for next-gen robot R&D?

We are working on hardware acceleration of MPC to make it applicable to very high-frequency control loops and want to make certain that this is an active research community before we invest in contributing to this library.

markusgft, 2021-08-02:
Hi, we occasionally fix bugs or fill in small new features. Major Development is currently not happening. The repo is sparsely maintained, but can serve you as inspiration!

Ravenwater, 2021-08-02:
@markusgft thank you for the guidance. For hw accelerated MPC co-development, where/who/what should we be connecting to?

markusgft, 2021-08-02:
From our side, we currently unfortunately cannot support you in detail regarding this feature request. Contributions in this direction would of course always be welcome!